moonlight-stream/moonlight-embedded #681: Cannot navigate through Big Picture menu, controller detected but no input.

controlstudent, 2018-08-08:
**_NVidia Geforce Experience version:_** 3.14.1.48
**Moonlight Embedded version:** 2.4.6
**Moonlight Embedded source:** [https://github.com/TechWizTime/moonlight-retropie](https://github.com/TechWizTime/moonlight-retropie)
**Moonlight Embedded running on:** Raspberry Pi 3 Model B V1.2
**Moonlight Embedded running on distribution:** Raspbian GNU/Linux 9 (stretch)

**Verbose output `-verbose` of Moonlight Embedded:**
 $ moonlight stream -720 -verbose
Moonlight Embedded 2.4.6 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.0.17...
NVIDIA GeForce GTX 670, GFE 3.14.1.48 (gs_04_08_24506279, 7.1.408.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Detected Xbox 360 Wireless Receiver (XBOX) (030000005e040000a102000007010000) on                               /dev/input/event0 as Xbox 360 Wireless Receiver (XBOX)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Unmapped button: 706
Unmapped button: 706
Unmapped button: 706
Unmapped button: 706
Unmapped button: 706
Unmapped button: 706
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
Once in BP mode on Steam, I should be able to navigate through the menus.

**What happens instead of that?**
I cannot navigate at all within BP mode, controller inputs do nothing. 

The controller and Wan Hung Lo XBox receiver I use work just fine, so I know that is not the problem. The keyboard and mouse on the host PC has control of BP. I see that I have some unmapped buttons in the -verbose. From the dozens of posts regarding controller mapping I am confident I can fix that. However, the primary problem (controller inputs do nothing) I believe is different.

welly321, 2018-08-08:
See issue [679](https://github.com/irtimmer/moonlight-embedded/issues/679) 
It seems the repo needs updating. If you compile it yourself, you can get it working. I haven't tried as I am awaiting a new release but others have had success.

irtimmer, 2018-08-11:
Duplicate of #679

closed 2018-08-11 by irtimmer