moonlight-stream/moonlight-embedded #684: Can't map ... controller not working correctly

Bjorkquist, 2018-08-17:
Please provide the following info.

**_NVidia Geforce Experience version: Same result for both latest and v3.13.1.30
**Moonlight Embedded version: 2.4.7
**Moonlight Embedded source: raspbian stretch branch via apt-get
**Moonlight Embedded running on: Raspberry Pi 3 Model B+
**Moonlight Embedded running on distribution: Raspbian

**Verbose output `-verbose` of Moonlight Embedded:
Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.1.126...
NVIDIA GeForce GTX 1070, GFE 3.13.1.30 (gs_04_05_23698325, 7.1.405.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Detected Microsoft X-Box 360 pad (030000005e0400008e02000014010000) on /dev/input/event0 as X360 Controller
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done

**What is the expected result?**
Gamepad working correctly, being able to use triggers and Dpad in games.

**What happens instead of that?**
Input is incorrect in games. Dpad and triggers are not working. This issue persists on all input devices tested. (Xbox360 wired controllers [3 different ones], Xbox One Controller and 8bitdo SF30 Pro)

BlueGeckoO, 2018-08-21:
I had a problem with my xbox one s pad not mapped correctly (not fixed in 2.4.7) and I just added a line into the gamecontrollerdb.txt with my controller id and copied the configuration of another xbox one controller (hoping it worked .. it did hopefully :p) 

What you can try:
- check if your controller id is in the file : `cat /usr/share/moonlight/gamecontrollerdb.txt | grep 030000005e0400008e02000014010000`
If nothing pops up, it just means there isn't a mapping for your controller with that ID.

- look if there is already a mapping that might work : `cat /usr/share/moonlight/gamecontrollerdb.txt | grep 360`

I found that one : 

> 030000005e0400008e02000062230000,Microsoft X-Box 360 pad,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,

- Now you can copy that mapping in the end of the file and change the controller id and the controller name with your own. Like that :

> 030000005e0400008e02000014010000,Microsoft X-Box 360 pad,a:b0,b:b1,back:b6,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b8,leftshoulder:b4,leftstick:b9,lefttrigger:a2,leftx:a0,lefty:a1,rightshoulder:b5,rightstick:b10,righttrigger:a5,rightx:a3,righty:a4,start:b7,x:b2,y:b3,

- Note: You can also create a mapping from scratch but have to figure out if "a" button is "b0", "b" button is "b1", "back" is "b6", etc.  If someone knows how to find the corrects mapping for each button, trigger, stick, I'm interested to know how !

Bjorkquist, 2018-08-21:
Oh, it seems like moonlight is trying to map dpad as axis. Don't know why though.

Bjorkquist, 2018-08-22:
Ok, does your gamepad work the way it should? I can navigate Big Picture but can't play games as the dpad and triggers are unmapped.

My gamepad is in the file and it's correctly mapped. I don't think this has to do with gamescontrollerdb.txt. My guess is that this is related to drivers, possibly a compatibility issue with udev? I think evdev is replaced with libdev in stretch, might be the reason but i suspect udev is responsible as it maps dpad as axis. It's trying to assign the left analog mappings to the dpad.

Are you running retropie as well? Really want this to work, might try with clean raspbian and perhaps compile moonlight-embedded from master branch.

BlueGeckoO, 2018-08-22:
The gamepad works as it should in steam big picture and any game I launch. I tested the last version of moonlight-embedded from the repositories (2.4.7) and before that, I compiled the 2.4.6. Turned out to be ok too once I resolved my problem.

I'm running a clean raspbian stretch, with retropie installed on it. Retropie works, I can play snes emulated games with the same controller. Also added a small script to start moonlight from emulationstation and it works too, starts big picture then any game in 1080p 60fps.

But took a while for figuring out how making all that to work. Now I'm struggling with 5.1 surround sound :-p

Bjorkquist, 2018-08-30:
Ok, so I actually got everything working fine. Still get the "can't map" postings /error but dpad and triggers are mapped correctly.

I simply installed raspbian stretch (lite) and then installed moonlight and retropie. Had installed RetroPie to SD via image before and that seems to be incompatible with moonlight for some reason.

irtimmer, 2018-08-30:
The 'can't map' are because some of the mappings (some uncommon gamepads) contains '-' or '+' mappings which are not supported by Moonlight and will be removed in the next release. Therefore these error messages have nothing to do with the controller problems

closed 2018-08-30 by irtimmer