moonlight-stream/moonlight-embedded #688: PS4 DUALSHOCK with  USB Wireless Adaptor not working

valievdmitrii, 2018-09-11:
Please provide the following info.

**_NVidia Geforce Experience version: 3.14.1.48
**Moonlight Embedded version:  2.4.7 (PI;ALSA;EMBEDDED)
**Moonlight Embedded source: http://archive.itimmer.nl/raspbian/moonlight/pool/main/m/moonlight-embedded/
**Moonlight Embedded running on:** Raspberry Pi 3 model B
**Moonlight Embedded running on distribution:** Raspbian

Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to *.*.*.*

NVIDIA GeForce GTX 970, GFE 3.14.1.48 (gs_04_08_24506279, 7.1.408.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Detected Sony Interactive Entertainment DUALSHOCK®4 USB Wireless Adaptor (030000004c050000a00b000011810000) on /dev/input/event2 as PS4 Controller
Detected Sony Interactive Entertainment DUALSHOCK®4 USB Wireless Adaptor Touchpad (030000004c050000a00b000011810000) on /dev/input/event0 as PS4 Controller
Detected Sony Interactive Entertainment DUALSHOCK®4 USB Wireless Adaptor Motion Sensors (030000004c050000a00b000011810000) on /dev/input/event1 as PS4 Controller

Gamepad not working ;(

patrickemuller, 2018-10-05:
You can map the gamepad and add to `/usr/share/moonlight/gamepadcontroller.txt` manually.
(please, check the name of the file first, don't remember for sure)
You can run
`moonlight map -input /dev/input/event0` and after mapping the keys just copy the result text inside the gamepad file I say above.

valievdmitrii, 2018-12-20:
thanks fot

> You can map the gamepad and add to `/usr/share/moonlight/gamepadcontroller.txt` manually.
> (please, check the name of the file first, don't remember for sure)
> You can run
> `moonlight map -input [...]

thanks! it is help for me
I edited the file, add to /usr/share/moonlight/gamepadcontroller.txt identifier of my gamepad to the existing settings and it all worked

closed 2018-12-20 by valievdmitrii