moonlight-stream/moonlight-embedded #690: Platform 'auto' not found

XJIOP, 2018-09-17:
tinkerboard with rockchip

```
# cmake ../
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.1.3")
-- Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so
-- Could NOT find Broadcom (missing:  BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY)
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
-- Found Rockchip: /usr/include/rockchip
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Checking for module 'libevdev'
--   Found libevdev, version 1.5.6
-- Checking for module 'libudev'
--   Found libudev, version 232
-- Checking for module 'sdl2>=2.0.4'
--
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'x11'
--   Found x11, version 1.6.4
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libpulse-simple'
--   Found libpulse-simple, version 10.0
-- Checking for module 'libcec>=4'
--
-- Checking for module 'egl'
--   Found egl, version 7.10
-- Checking for module 'glesv2'
--   Found glesv2, version
-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.52.1")
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found suitable version "1.1.0f", minimum required is "1.0.2")
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.2.0")
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.6.32
-- Checking for module 'libenet'
--   Found libenet, version 1.3.12
-- Configuring done
WARNING: Target "moonlight-rk" requests linking to directory "/usr/lib/arm-linux-gnueabihf".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
-- Build files have been written to: /home/linaro/src/moonlight-embedded/build
```

```
# moonlight stream -app Steam
Platform 'auto' not found
```

```
# moonlight stream -app Steam -platform fake
No mapping available for RC for dw_hdmi (1e000000524320666f722064775f6800) on /dev/input/event1
No mapping available for gpio-keys (19000000010000000100000000010000) on /dev/input/event3
No mapping available for rockchip,miniarm-codec Headphones (00000000726f636b636869702c6d6900) on /dev/input/event2
```

```
# cd /dev/input/by-path && ls
platform-ff500000.usb-usb-0:1:1.255-event  
platform-ff980000.hdmi-event  
platform-gpio-keys-event  
platform-sound-simple-card-event
```

how to fix it?

ccuono, 2018-12-02:
I have pretty much the same exact situation going on for me; here is my report:

**NVidia Geforce Experience version:** 3.16.0.122
**Moonlight Embedded version:** 2.4.7
**Moonlight Embedded source:** _Compiled from source_
**Moonlight Embedded running on:** Rock64 4G Memory (Rockchip RK3288)
**Moonlight Embedded running on distribution:** Ubuntu 18.04 Bionic

**Verbose output `-debug` of Moonlight Embedded:**
Any identifiers I replaced with zzz, xxx, yyy etc...
```
moonlight stream -debug -4k -fps 60 -app mstsc.exe 192.168.1.2
Moonlight Embedded 2.4.7 (ROCKCHIP;ALSA;PULSE;EMBEDDED)
Connect to 192.168.1.2...
Request https://192.168.1.2:47984/serverinfo?uniqueid=zzzzz&uuid=yyyyy
Response:
<?xml version="1.0" encoding="UTF-16"?><root protocol_version="0.1" query="serverinfo" status_code="200" status_message="OK"><AuthenticationType>1</AuthenticationType><ConnectionState></ConnectionState><CurrentClient>0</CurrentClient><GfeVersion>3.16.0.122</GfeVersion><GsVersion>gs_04_11_25137997</GsVersion><HttpsPort>47984</HttpsPort><LocalIP>192.168.1.2</LocalIP><LocalIPs><Address>192.168.1.2</Address></LocalIPs><LocalIPs><Address>192.168.2.1</Address></LocalIPs><LoginState>2</LoginState><MaxLumaPixelsH264>737235008</MaxLumaPixelsH264><MaxLumaPixelsHEVC>589622848</MaxLumaPixelsHEVC><Mode>0</Mode><PairStatus>1</PairStatus><ServerCapability>3895</ServerCapability><ServerCodecModeSupport>3843</ServerCodecModeSupport><ServerColorSpaceSupport>2</ServerColorSpaceSupport><SupportedDisplayMode><DisplayMode><Height>2160</Height><RefreshRate>60</RefreshRate><Width>3840</Width></DisplayMode><DisplayMode><Height>2160</Height><RefreshRate>30</RefreshRate><Width>3840</Width></DisplayMode><DisplayMode><Height>1080</Height><RefreshRate>60</RefreshRate><Width>1920</Width></DisplayMode><DisplayMode><Height>1080</Height><RefreshRate>30</RefreshRate><Width>1920</Width></DisplayMode><DisplayMode><Height>720</Height><RefreshRate>60</RefreshRate><Width>1280</Width></DisplayMode><DisplayMode><Height>720</Height><RefreshRate>30</RefreshRate><Width>1280</Width></DisplayMode></SupportedDisplayMode><accountId>xxxx</accountId><appversion>7.1.411.0</appversion><currentgame>0</currentgame><gamelistid>zzz</gamelistid><gputype>GeForce GTX 1050 Ti</gputype><hostname>Z820-GAMING</hostname><mac>www</mac><numofapps>0</numofapps><resyncSuccessful>0</resyncSuccessful><state>MJOLNIR_STATE_SERVER_AVAILABLE</state><uniqueid>vvv</uniqueid></root>

NVIDIA GeForce GTX 1050 Ti, GFE 3.16.0.122 (gs_04_11_25137997, 7.1.411.0)
Platform Unknown
Platform 'auto' not found

```

**What is the expected result?**
Stream from PC (mstsc.exe) to moonlight-embedded

**What happens instead of that?**
Nothing. All other platforms I get a message `Platform 'platform' not found` (I've tried what's listed in the documents like x11, x11_vdpau, etc) except for when I use fake.

**I get the following behavior if I try `-platform fake`:**

Nothing gets displayed, I can see on the remote machine's monitor that I can control the mouse/keyboard but there is no display coming through on the Rock64 (it just looks like a frozen from from where I launched the command). I can still quit the stream via ctrl + shift + alt + q.

**cmake log:**
```
cmake ../
-- The C compiler identification is GNU 7.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ALSA: /usr/lib/aarch64-linux-gnu/libasound.so (found version "1.1.3") 
-- Found Opus: /usr/lib/aarch64-linux-gnu/libopus.so  
-- Could NOT find Broadcom (missing: BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY) 
-- Could NOT find Freescale (missing: FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
-- Could NOT find Amlogic (missing: AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY) 
-- Found Rockchip: /usr/local/include/rockchip  
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'libevdev'
--   Found libevdev, version 1.5.8
-- Checking for module 'libudev'
--   Found libudev, version 237
-- Checking for module 'sdl2>=2.0.4'
--   No package 'sdl2' found
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'x11'
--   Found x11, version 1.6.4
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libpulse-simple'
[... 4 lines cut ...]
--   Found egl, version 18.0.5
-- Checking for module 'glesv2'
--   Found glesv2, version 18.0.5
-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CURL: /usr/lib/aarch64-linux-gnu/libcurl.so (found version "7.58.0") 
-- Found OpenSSL: /usr/lib/aarch64-linux-gnu/libcrypto.so (found suitable version "1.1.0g", minimum required is "1.0.2") 
-- Found EXPAT: /usr/lib/aarch64-linux-gnu/libexpat.so (found version "2.2.5") 
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.7
-- Checking for module 'libenet'
--   Found libenet, version 1.3.12
-- Configuring done
WARNING: Target "moonlight-rk" requests linking to directory "/usr/lib/aarch64-linux-gnu".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
-- Build files have been written to: /home/rock64/source/moonlight-embedded/build
```

irtimmer, 2019-04-30:
bfade0686b0b00f31626daa7fea98dd221989e07 should correctly link libmoonlight-rk to libdrm

closed 2019-04-30 by irtimmer