moonlight-stream/moonlight-embedded #692: No game-stream possible (only windows desktop)

Crunky, 2018-09-24:
Hi,

i installed moonlight to raspberry Pi3+ with Debian Stretch. If i open a stream to the host, it is not possible to start a game or open Steam (in established stream it closes automatic). I only see the Windows Desktop every time.

RaspberryPI3+
Raspbian Stretch
GeForce_Experience_v3.15.0.164
GTX 1050 (411.63)
Disabled Firewall 
Moonlight installation (https://github.com/irtimmer/moonlight-embedded/wiki/Packages)
Pairing with Shield OK
"moonlight list" find steam app and also installed games
try with "stream IP" and "stream -app Steam"

Did i miss something?

Regards
Marco

patrickemuller, 2018-10-05:
besides opening the Steam, your desktop is streaming fine? Or you have some freeze or lag on the commands?

Crunky, 2018-10-06:
The connection sometimes lost or raspberry freeze ....  i bought steamlink also now.

mueller-werner, 2018-10-23:
same issue here.  i fixed it by removing my second screen on the Streaming Host Machine.

emuchicken, 2018-11-18:
Try turning off overlay on geforce experience.  Once that was off I could see the game streaming no problem.