moonlight-stream/moonlight-embedded #696: Home (Guide) and Select (Back) button maps as dpad up on Xbox One Wireless, works fine on X360 Wired

sondregronas, 2018-10-14:
**_NVidia Geforce Experience version:_** 3.15.0.164
**Moonlight Embedded version:** 2.4.7
**Moonlight Embedded source:** compiled from source
**Moonlight Embedded running on:** Raspberry Pi
**Moonlight Embedded running on distribution:** Raspbian, RetroPie OS

**Verbose output `-verbose` of Moonlight Embedded:**
Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
Connect to 192.168.10.240...
NVIDIA GeForce GTX 1080 Ti, GFE 3.15.0.164 (gs_04_09_24784787, 7.1.409.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Detected Xbox Wireless Controller (050000005e040000fd02000003090000) on /dev/input/event2 as Xbox One Wireless Controller
Detected Microsoft X-Box 360 pad (030000005e0400008e02000014010000) on /dev/input/event1 as X360 Controller
Detected Microsoft X-Box 360 pad (030000005e0400008e02000014010000) on /dev/input/event0 as X360 Controller
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
Xbox and Select button on Xbox One Wireless controllers work as separate inputs, to access the Steam Big Picture Menu

The gamecontrollerdb.txt looks correct to me:
`050000005e040000fd02000003090000,Xbox One Wireless Controller,a:b0,b:b1,back:b15,dpdown:h0.4,dpleft:h0.8,dpright:h0.2,dpup:h0.1,guide:b16,leftshoulder:b6,leftstick:b13,lefttrigger:a5,leftx:a0,lefty:a1,rightshoulder:b7,rightstick:b14,righttrigger:a4,rightx:a2,righty:a3,start:b11,x:b3,y:b4,`

**What happens instead of that?**
Xbox button & Select button give the same input as dpup (h0.1) when running steam. In RetroPie they are separate

I've tried: Running without xboxdrv, xpad. Running with either or both at the same time. I've also tried remapping it from the Steam menu but I get input dpup h0.1

TheUntouchable, 2018-10-21:
Have the exact same problem with my xbox one wireless controller:
moonlight stream 192.168.1.100 -verbose

Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
Connect to 192.168.1.100...
NVIDIA GeForce GTX 980 Ti, GFE 3.15.0.186 (gs_04_09_24784787, 7.1.409.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Detected Xbox Wireless Controller (050000005e040000fd02000003090000) on /dev/input/event2 as Xbox One Wireless Controller
Detected Xbox Gamepad (userspace driver) (0000000058626f782047616d65706100) on /dev/input/event1 as Xbox Gamepad (userspace driver)

JeroenM1, 2018-10-27:
Having the same issue.
Using the xbox 360 wireless wich is working fine until it`s resolved.

maltsevda, 2018-11-03:
+1. Having the same issue. Xbox One S Wireless Controller. Firmware version: 3.1.1221.0.

pokemonw2, 2018-11-27:
Hey guys i found a solution that works for me (Raspberry pi 3, Retropie)
I'll try to broadly outline the steps here:

1. Navigate to /home/pi/RetroPie-Setup
2. Fire up the retropie start up script `sudo sh retropie_setup.sh`
3. Go to manage packages, then driver, and then xpad
4. Remove xpad
5. Follow the instructions at https://github.com/atar-axis/xpadneo/ to install xpadneo and pair with controller
6. Reboot
7. Now you will have to add a mapping script for the controller. While i can't guarantee that your script will be the same as mine, you will have to add an extra line in /usr/local/share/moonlight/gamecontrollerdb.txt 
or 
/usr/share/moonlight/gamecontrollerdb.txt

Here's my script entry that worked for my controller: 

`050000005e040000fd02000030110000,Xbox Wireless
Controller,a:b0,b:b1,x:b2,y:b3,start:b7,guide:b8,back:b6,leftstick:b9,rightstick:b10,dpup:h0.9,dpleft:h0.8,dpdown:h0.12,dpright:h0.2,leftx:a0,lefty:a1,rightx:a3,righty:a4,lefttrigger:a2,righttrigger:a5,leftshoulder:b4,rightshoulder:b5,platform:Linux`

8. Reboot and see if it works

KiralyCraft, 2020-05-14:
I can confirm on my RB+ with Raspbian, that removing "xboxdrv" (apt-get remove xboxdrv) and following the installation steps from  https://github.com/atar-axis/xpadneo/ fixed the issue. Now works as expected

SalehCoder, 2020-08-03:
> I can confirm on my RB+ with Raspbian, that removing "xboxdrv" (apt-get remove xboxdrv) and following the installation steps from https://github.com/atar-axis/xpadneo/ fixed the issue. Now works as expected with the default mapping that came with moonlight-embedded

Worked perfectly! Thank you!

commit 4b41692f (2021-08-08): Fix handling of buttons outside the normal joystick range

See https://github.com/libsdl-org/SDL/commit/aa03b9d7afedefddda4e7f1575d87865ee5f3d93

Fixes #788
Fixes #696
Fixes #675
Fixes #693
Fixes #642
closed 2021-08-08 by cgutman