moonlight-stream/moonlight-embedded #705: GFE 3.16 not supported

jesusvallejo, 2018-11-20:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.16
**Moonlight Embedded version:** Moonlight Embedded 2.4.7
**Moonlight Embedded source:** _https://github.com/irtimmer/moonlight-embedded/releases/download/v2.4.7/moonlight-embedded-2.4.7.tar.xz_
**Moonlight Embedded running on:** _Raspberry Pi 3b_
**Moonlight Embedded running on distribution:** _Raspbian  stretch_

**Verbose output `-verbose` of Moonlight Embedded:**

pi@raspberrypi:~ $ moonlight stream -1080 -fps 60 -app windows
Searching for server...
Connect to 192.168.1.51...
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start A failed: 11
failed: 11
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
windows being mirrored

**What happens instead of that?**
black screen

GFE 3.15 works as expected , GFE 3.16 released monday 19th Nov 2018 makes the display go black and nothing else happens, no video, no audio.

cgutman, 2018-11-20:
The fix for common-c is in https://github.com/moonlight-stream/moonlight-common-c/commit/8665d806978d2cba9c9040183a3686f4d87d402c and it has shipped on PC, iOS, and Android clients without reported issues.

Tony322, 2018-11-20:
Can confirm. Running the same stuff on my RPI 3. Running Raspbian and using same launch options but -app Steam instead. I also have GFE 3.16.0.122.
"moonlight stream -1080 -fps 60 -app Steam"

JailSeed, 2018-11-20:
> The fix for common-c is in [moonlight-stream/moonlight-common-c@8665d80](https://github.com/moonlight-stream/moonlight-common-c/commit/8665d806978d2cba9c9040183a3686f4d87d402c) and it has shipped on PC, iOS, and Android clients without reported issues.

But how to update this file on RPI3?

jesusvallejo, 2018-11-21:
> The fix for common-c is in [moonlight-stream/moonlight-common-c@8665d80](https://github.com/moonlight-stream/moonlight-common-c/commit/8665d806978d2cba9c9040183a3686f4d87d402c) and it has shipped on PC, iOS, and Android clients without reported issues.

yeah i found that while searching for this issue , but more than an issue is a warning for irtimmer so he can release a version that uses the new commit. Also , dont know how all this works but could not find moonlight-stream/moonlight-common-c file in embedded so for the moment the only solution is to rollback to GFE 3.15

5schatten, 2018-11-21:
@jesusvallejo 
The file is located in the third-party subfolder so `moonlight-embedded/third_party/moonlight-common-c/src`

I've tested the patch https://github.com/moonlight-stream/moonlight-common-c/commit/8665d806978d2cba9c9040183a3686f4d87d402c at my build and it works as far as I can tell.

My settings:

```
MOONLIGHT_ARGS=stream -verbose -app Steam
Moonlight Embedded 2.4.7 (CEC;SDL;ALSA;PULSE)
Searching for server...
Connect to 192.168.1.10...
NVIDIA GeForce GTX 1060 6GB, GFE 3.16.0.122 (gs_04_11_25137997, 7.1.411.0)
Platform SDL2 (software decoding)
Stream 1920 x 1080, 60 fps, 20000 kbps
```

JailSeed, 2018-11-21:
> @jesusvallejo
> The file is located in the third-party subfolder so `moonlight-embedded/third_party/moonlight-common-c/src`
> 
> I've tested the patch [moonlight-stream/moonlight-common-c@8665d80](https://github.com/moonlight-stream/moonlight-common-c/commit/8665d806978d2cba9c9040183a3686f4d87d402c) at my build and it works as far as I can tell.
> 
> My settings:
> 
> ```
> MOONLIGHT_ARGS=stream -verbose -app Steam
> Moonlight Embedded 2.4.7 (CEC;SDL;ALSA;PULSE)
> Searching for server...
> Connect to 192.168.1.10...
> NVIDIA GeForce GTX 1060 6GB, GFE 3.16.0.122 (gs_04_11_25137997, 7.1.411.0)
> Platform SDL2 (software decoding)
> Stream 1920 x 1080, 60 fps, 20000 kbps
> ```

Yeah, thank you, thats just worked for me (RPI3, OSMC)

fl0wzr, 2018-11-22:
> @jesusvallejo
> The file is located in the third-party subfolder so `moonlight-embedded/third_party/moonlight-common-c/src`
>
> I've tested the patch [...]

Thank you for providing your already gained knowledge! But my question is how I update the external moonlight-common-c library to this specific patch?

5schatten, 2018-11-22:
@fl0wzr 
Well in my case I compile the whole stuff myself so I apply the patch and the buildsystem does the rest. If you rely on apt-get you have to wait until the repo is updated.

TyanColte, 2018-11-23:
Has anybody tested running moonlight-embedded with the -unsupported flag? Since us plebs who don't know how to compile the whole thing ourselves and rely on somebody who does to update the repo in a timely fashion can't stream right now do you think the -unsupported option would work?

hhromic, 2018-11-23:
I can confirm the issue on my side as well. The patch mentioned does fix it.
If you are using RetroPie, you can try the experimental scriptmodule I'm developing here:
https://retropie.org.uk/forum/topic/19872/moonlight-request-for-comments-and-testing

It will automatically build, install and configure moonlight in your RetroPie machine.
I included now the mentioned patch to fix compatibility with GFE 3.16

hhromic, 2018-11-23:
I sent a PR now against the moonlight embedded version of  `moonlight-common-c`:
https://github.com/irtimmer/moonlight-common-c/pull/1

Hopefully @irtimmer merges it soon and all should be back to normal.

fl0wzr, 2018-11-23:
> I sent a PR now against the moonlight embedded version of `moonlight-common-c`:
> [irtimmer/moonlight-common-c#1](https://github.com/irtimmer/moonlight-common-c/pull/1)
> 
> Hopefully @irtimmer merges it soon and all should be back to normal.

That's actually really great, thank you! Hopefully this will get merged into the main branch soon.

irtimmer, 2018-11-23:
Issue is fixed with commit 9ab10f0dab66f49e4e74000d7b51f64bdce25f8b

closed 2018-11-23 by irtimmer

hhromic, 2018-11-23:
Great, thanks @irtimmer !

fl0wzr, 2018-11-23:
Awesome, thank you! @irtimmer

hhromic, 2018-11-23:
Confirmed working! ;)  Thanks again.

fl0wzr, 2018-11-24:
Can't confirm, it's not working for me..

jesusvallejo, 2018-11-24:
Did you build it on your device or just reinstalled? you'll hav to compile it if you want it to work, and i confirm it works.
save the "first"as a script like install.sh and `sudo sh install.sh` , and it will install everything you need.

First: 
`INSTALL_PKGS=" libopus0 libexpat1 libasound2 libudev0 or libudev1 libavahi-client3 libcurl3 libevdev2 libenet7 rbp-userland-osmc libraspberrypi0  libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev libpulse-dev uuid-dev libenet-dev cmake gcc g++ libraspberrypi-dev  fakeroot debhelper "
for i in $INSTALL_PKGS; do
  sudo apt-get install -y $i
done
`
"Second" execute all those commands:

Second:
`git clone git@github.com:irtimmer/moonlight-embedded.git`

`cd moonlight-embedded`

`git submodule update --init --recursive`

`mkdir build
cd build/
cmake ../
make
make install`

`sudo ldconfig`

`moonlight stream -1080 -fps 60 -bitrate 8000`

and it should start streaming steam big picture.

fl0wzr, 2018-11-24:
@jesusvallejo 
Thank you for answering and with such detail! 

I just tried to updated my moonlight-embedded version (sudo apt-get update) but I guess it was just merged into the main branch but no new version was released? I'll give your approach a try! :-)

fl0wzr, 2018-11-24:
@jesusvallejo 
So I tried your approach... and it worked!! thank you a lot!👍 

But just for your interest, the script doesn't work. Following error appears: 
`pi@raspberrypi:~ $ sudo sh install.sh
install.sh: 2: install.sh: for: not found
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package done
`
Maybe some repository is faulty. Created all I needed manually.

posti85, 2018-11-25:
Thank you @jesusvallejo, now I've moonlight working!

I only have to point that the script you provided to install dependencies didn't work for me. Based on the dependencies from https://github.com/irtimmer/moonlight-embedded/wiki/Compilation I just executed:

```
sudo apt-get update
sudo apt-get install libopus0 libexpat1 libasound2 libudev1 libavahi-client3 libcurl3 libevdev2 libenet7 libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev libpulse-dev uuid-dev libenet-dev cmake gcc g++ fakeroot debhelper
```

Also, if you have OSMC:
```
sudo apt-get install rbp-userland-dev-osmc
```

Or if you have Raspbian:
```
sudo apt-get install libraspberrypi-dev
```

Then I followed the second part (cloning and compiling the project). Just point again that I had to:

```
sudo make install
```

(yep, with sudo)

And after all I had to

```
sudo shutdown -r now
```

To get it working.

One question: What should I do if the release gets updated and I want to use the executable from the repo instead of this compiled one?

Best regards.

jesusvallejo, 2018-11-27:
@fl0wzr you are welcome, the script should work as I used it and copy/paste it in here, I'll update it for future reference.
@posti85 thanks for the update, and also confirm that is sudo make install , I forgot to write it in here, if you sudo ldconfig you shouldn't have to reboot neither shutdown, but it's another way to do it.

As for your question related to the update, I have no knowledge to answer it but you shouldn't have any trouble to apt remove and install it from the repository.

iTaybb, 2018-11-27:
I've compiled the latest version from git, however I'm still getting the error:
`Moonlight Embedded 2.4.7 (CEC;PI;SDL;X11;VDPAU;VAAPI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 10.100.102.7...
NVIDIA GeForce GTX 1060 3GB, GFE 3.16.0.122 (gs_04_11_25137997, 7.1.411.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 11
failed: 11
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

`

posti85, 2018-11-27:
@iTaybb, did you uninstalled the other package? Did the compilation work? The error is throwing seems to be the same so it's being use the previous version. Make sure you are using the executable you have compiled.

iTaybb, 2018-11-27:
> 
> 
> @iTaybb, did you uninstalled the other package? Did the compilation work? The error is throwing seems to be the same so probably it's being used the previous version. Make sure you are using the executable you have compiled.

Thank you. I didn't uninstall the old package beforehand. Now it's all working.

MrScruff300, 2018-12-07:
Hi all,

I am having the same issues as above since updating GeForce and unfortunately i’m not confident enough to update by compiling myself (even with instructions above). I launch moonlight through the Retropie main screen, I followed instructions to create an icon and launch script. Will the release in the repository be updated to correct this error as i will need to be able to update through audio apt get update? Cheers

MrScruff300, 2018-12-07:
Meant sudo apt-get update

posti85, 2018-12-07:
@MrScruff300, here: https://github.com/irtimmer/moonlight-embedded/issues/716#issuecomment-445205242 you have detailed instructions to compile it

devolt5, 2019-02-10:
Here's what worked for me:
1. Fresh installation of RetroPie 4.4 with Raspian 9 - Stretch
2. Followed instructions on: [#716](https://github.com/irtimmer/moonlight-embedded/issues/716#issuecomment-445205242)

If you use the script from https://github.com/TechWizTime/moonlight-retropie replace "moonlight jessie main" with "moonlight stretch main"

Now it work's again with GFE 3.16.0.140

McNugget6750, 2019-02-17:
I'm trying to follow the instructions as I get the error described in 

However, when I try to run `git submodule update --init` I get an error that I can't get rid or. Therefore I can't build from sources. Any ideas?

 git submodule update --init
Submodule 'common' (https://github.com/irtimmer/moonlight-common-c.git) registered for path 'third_party/moonlight-common-c'
Cloning into '/home/pi/moonlight-embedded/third_party/moonlight-common-c'...
error: no such remote ref b92e4884aabd3af865c512a20366560b337619d3
Fetched in submodule path 'third_party/moonlight-common-c', but it did not contain b92e4884aabd3af865c512a20366560b337619d3. Direct fetching of that commit failed.

McNugget6750, 2019-02-17:
and if I ignore the error and try to build:

git submodule update --init
error: no such remote ref b92e4884aabd3af865c512a20366560b337619d3
Fetched in submodule path 'third_party/moonlight-common-c', but it did not contain b92e4884aabd3af865c512a20366560b337619d3. Direct fetching of that commit failed.
pi@raspberrypi:~/moonlight-embedded $ mkdir build
pi@raspberrypi:~/moonlight-embedded $ cd build/
pi@raspberrypi:~/moonlight-embedded/build $ cmake ../
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.1.3")
-- Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so
-- Found Broadcom: /opt/vc/include
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
-- Could NOT find Rockchip (missing:  ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Checking for module 'libevdev'
--   Found libevdev, version 1.5.6
-- Checking for module 'libudev'
--   Found libudev, version 232
-- Checking for module 'sdl2>=2.0.4'

-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'x11'
--   No package 'x11' found
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libpulse-simple'
--   Found libpulse-simple, version 10.0
-- Checking for module 'libcec>=4'

-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.52.1")
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found suitable version "1.1.0j", minimum required is "1.0.2")
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.2.0")
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.6.32
-- Checking for module 'libenet'
--   Found libenet, version 1.3.12
You have called ADD_LIBRARY for library moonlight-common without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
CMake Error: CMake can not determine linker language for target: moonlight-common
CMake Error: Cannot determine link language for target "moonlight-common".
-- Generating done
-- Build files have been written to: /home/pi/moonlight-embedded/build
pi@raspberrypi:~/moonlight-embedded/build $ make
Scanning dependencies of target moonlight-common
make[2]: *** No rule to make target 'libgamestream/CMakeFiles/moonlight-common.dir/build'.  Stop.
CMakeFiles/Makefile2:163: recipe for target 'libgamestream/CMakeFiles/moonlight-common.dir/all' failed
make[1]: *** [libgamestream/CMakeFiles/moonlight-common.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

I

marcusobrien, 2019-11-25:
Build instructions work perfectly on latest Raspbian (Buster) with Raspberry Pi 4. Can get 1080 with 60 FPS. Some textures dont seem to come through though on Outer World, and Witcher 3 (only games I've tried), streaming PC desktop also seems to work. Not sure what causes this. I will try a LAN cross over cable between PC and Pi, bypass my router, see if its LAN bandwidth related. Incidentally increasing the memory for GPU on the Pi makes no difference (1GB setting causes Kernel deadlock on boot, 256MB seems to be enough).
Specs are Pi 4 is 4GB model, PC 64GB DDR 4 3200, AMD Ryzen 3900x, NVidia 1650 GTX