moonlight-stream/moonlight-embedded #709: Starting control stream...Start B failed: 11

randomxsk8, 2018-11-24:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.16.0.122
**Moonlight Embedded version:** 2.4.7
**Moonlight Embedded source:** https://github.com/joelghill/retromoonlight
**Moonlight Embedded running on:** Raspberry Pi 3
**Moonlight Embedded running on distribution:** Retropie

**Verbose output `-verbose` of Moonlight Embedded:**
`
Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
Connect to 192.168.1.134...
NVIDIA GeForce GTX 1060, GFE 3.16.0.122 (gs_04_11_25137997, 7.1.411.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 11
failed: 11
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done`

**What is the expected result?**
That moonlight words

**What happens instead of that?**
It stopped after Starting control stream...Start B failed: 11 and it cleans out

I also tried with xbox one wired and bluetooth controller. Anyway i saw that is a[ geforce version](https://github.com/irtimmer/moonlight-embedded/issues/313) problem

petursig, 2018-11-24:
I am having the same issue since 21st Nov. I do not have any specific controller mapped.

fl0wzr, 2018-11-24:
Same issue! 

Also mentioned in the already closed issue #705

randomxsk8, 2018-11-24:
SOLVED:
I updated moonlight from this [script ](https://retropie.org.uk/forum/topic/19872/moonlight-request-for-comments-and-testing) posted in #705 by hhromic.
I don't know if you had controller mappig problems.. I solved using the [SDL2 Gamepad Tool](http://www.generalarcade.com/gamepadtool/) generate a txt with my mapped pad and i linked its path in global.conf located at /opt/retropie/configs/all/moonlight/global.conf
It works perfectly :)
Thanks to everyone for your support!

posti85, 2018-11-25:
Any news about a new release in the official repo? I really don't know how to compile it (the instructions from the wiki doesn't work for me)...

Coastline23, 2018-11-27:
Just a point from someone who recently downloaded Moonlight.  

Could the "Requirements" on the front page that state "Geforce Experience 2.1.1 or higher" be edited to say "Geforce Experience 2.1.1 - 3.15 (not working from 3.16)".

jesusvallejo, 2018-11-27:
> Just a point from someone who recently downloaded Moonlight.
> 
> Could the "Requirements" on the front page that state "Geforce Experience 2.1.1 or higher" be edited to say "Geforce Experience 2.1.1 - 3.15 (not working from 3.16)".

well it works, you just have to compile it , a new release & a repository update would be grate , but it  works.

posti85, 2018-11-27:
If someone have the knowledge of generating a .deb file, and generates and uploads it to some hosting, would be great for those people who don't know how to compile. 

I recorded a tutorial in YouTube explaining how to install moonlight-embedded in OSMC, and I receive multiple requests to explain them how to make it work with the current issue...