moonlight-stream/moonlight-embedded #711: Bad Audio Latency

Aroobn, 2018-11-28:
Pretty bad Audio delay with embedded, running GeForce 3.15, Latest Moonlight. I have the internet to run it fine but still audio delay on 1080p, 720 and 480p

kurac338, 2018-12-18:
What is your bitrate set to? I use moonlight primarily on LAN, with a wired RPI 3.
I used 30 000 bitrate, on 1080p60 for a couple of months without a problem.
Recently I got a bigger / better TV and bumped the bitrate even higher to reduce the artifacts. Now it is set to 40 000, still without issues (video, audio and input all in sync).
Also not sure if it could be related, but monitor your Pi's temperature and CPU load.
I have a big heatsink on it (Pi 3 B, non +), and it has greatly helped with the temperatures (approx. -15 C).

For remote usage you need to play primarily with your resolution + framerate + bitrate that will be within the range of your particular upload speed. Also you might need to adjust your firewall settings not to interfere with GFE's ports / application.

noblackthunder, 2019-08-02:
i have the same issue ... streamed minecraft in 144p at 40mbps from my main computer to my Surface pro 6 (core i7 version) via wifi (i am right beside the router) 

Video was perfect but audio seemed to be delayed around a second 

anywhere i can find logs and attach them here maybe ?

kurac338, 2019-08-02:
> i have the same issue ... streamed minecraft in 144p at 40mbps from my main computer to my Surface pro 6 (core i7 version) via wifi (i am right beside the router)
> 
> Video was perfect but audio seemed to be delayed around a second
> 
> anywhere i can find logs and attach them here maybe ?

Hi, your issue seems to me linked to another moonlight project made for x86 platforms. Moonlight-embedded is made for ARM architecture (single board computers like Raspberry Pi).

Please refer to a corresponding project here: https://github.com/moonlight-stream
It is likely that you are using moonlight-qt (at least you should be using that since others are discontinued), so file an issue there if there isn't already a related report.

noblackthunder, 2019-08-02:
ohh sorry .. thanks !