moonlight-stream/moonlight-embedded #712: how to delete all previous configurations

eglische, 2018-11-29:
Hello!

Running Retropie/Rapbarry3b:
Got moonlight installed over a messed up script from a tutorial i tried and yeah, it points to a wrong ip.

But i can't get the wrong IP address sorted out. I want to reset everything.
I tried to search for a moonlight.conf but there is nothing set, also wiped the .cache/moonlight folder with the keys , then reinstall everything and than re-paired to the correct ip. Still, moonlight tries the wrong old ip adress, even after successful pairing.

So where would that ip be stored?

Thanks
Yeti

pokemonw2, 2018-11-29:
How do you launch moonlight?
If you do it by script, there's a chance that the wrong ip is actually in the script you're trying to launch moonlight with. If there's nothing in /usr/local/etc/moonlight.conf then maybe moonlight is just defaulting to the wrong ip at random and you will have to specify your ip using when launching moonlight.

eglische, 2018-11-29:
i checke the script already and also removed it. it inputs into $ip which i think would remain in the ram, but should be gone after reboot.

however, when i try to start "moonlight pair (corretiphere) -verbose that is what i get:
`Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
Connect to 192.168.1.199...
NVIDIA GeForce GTX 970, GFE 3.16.0.122 (gs_04_11_25137997, 7.1.411.0)
Please enter the following PIN on the target PC: 4557
Failed to pair to server: Failed to sign data
`

i deleted the keys but still get the same issue.
If i try to unpair, then the old ip pops up. however, when i enter the ip while unpairing it works.
but still, after try to pair again, it spits out the same problem.
so i thought beside the ip address, the script could have been messing up something else as well, thats why i ask how to do a "clean" reinstall of everything

posti85, 2018-11-30:
Did ypu try moonlight list command? That will show connecting to IP...