moonlight-stream/moonlight-embedded #713: GFE 3.16 unsuported - GameStream Protocol changes?

Nodrev, 2018-12-01:
Please provide the following info.

**_NVidia Geforce Experience version: 3.16.0.122**
**Moonlight Embedded version: 2.4.7**
**Moonlight Embedded running on:** Raspberry Pi
**Moonlight Embedded running on distribution:** Raspbian Stretch

**Verbose output `-verbose` of Moonlight Embedded:**
```
moonlight stream 192.168.0.12 -1080 -app Steam -verbose
Moonlight Embedded 2.4.7 (PI;ALSA;EMBEDDED)
Connect to 192.168.0.12...
NVIDIA GeForce GTX 670, GFE 3.16.0.122 (gs_04_11_25137997, 7.1.411.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...Start B failed: 11
failed: 11
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
Error: cannot keep u
```

**What is the expected result?**
Launching Steam...

**What happens instead of that?**
The raspberrypi seem to hang: the raspbian desktop is visible (not Steam), no mouse control nor keyboard control.
But in fact, it's moonlight who have the hand, because mouse and keyboard controls comes back when firing moonlight's escape key sequence (ctrl+alt+shit+Q).
After that, I log on on the target PC hosting GFE using VLC, and saw that the "Launch Steam" command was processed correctly, as Steam was launched in big picture mode.
This appends since I updated my GFE version yesterday, everything was working fine with moonlight 2.4.7 and my old GFE installation (didn't remember the exact version but it was installed in may 2018).

Nodrev, 2018-12-01:
As a side note, the first time I execute the command, the error is:
```
Initializing input stream...done
Starting control stream...Start A failed: 11
failed: 11
```

And subsequents launches results in:
```
Initializing input stream...done
Starting control stream...Start B failed: 11
failed: 11
```

irtimmer, 2018-12-01:
Duplicate of #705

closed 2018-12-01 by irtimmer

Nodrev, 2018-12-01:
Compiled master, works like a charm, thanks again for pointing me out to the solution!