moonlight-stream/moonlight-embedded #715: Newest Geforce drivers (417.22) won't stream (398.26 works fine)

emuchicken, 2018-12-06:
Please provide the following info.

**_NVidia Geforce Experience version:_ 3.16.0.122 **
**Moonlight Embedded version: Moonlight Embedded 2.4.7**
**Moonlight Embedded source:** _repository
**Moonlight Embedded running on:** _Raspberry Pi
**Moonlight Embedded running on distribution:** retropie

**Verbose output `-verbose` of Moonlight Embedded:**
??

**What is the expected result?**
black screen, then streaming screen on TV
**What happens instead of that?**
black screen, then no streaming- even though Steam has loaded on the host machine.
Tested with old geforce drivers (398.26) and it works fine again.

hhromic, 2018-12-06:
Duplicate of #705 

@irtimmer maybe is a good time to bump the repository packages, otherwise users won't see the recent updates and will keep posting this issue over an over :)

hhromic, 2018-12-06:
@emuchicken also I see you are using RetroPie. Why not give it a try to the experimental module I'm proposing in <https://retropie.org.uk/forum/topic/19872/moonlight-request-for-comments-and-testing> ?
 It compiles moonlight from source so you will get the updated version that works with GFE 3.16.
Don't forget to clean/remove your current installation of moonlight.