moonlight-stream/moonlight-embedded #720: Unable to connect Raspberry Pi 2 to Windows 7 Desktop (RTSP: request failed 404)

hayleyhailstorm, 2018-12-16:
Please provide the following info.

**_NVidia Geforce Experience version:_** Version 3.16.0.140
**Moonlight Embedded version:** 2.1.2
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._ Raspberry Pi
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._ Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**
Moonlight runs

**What happens instead of that?**
Receive error message:
moonlight stream
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Can't map (reverse_z)
Can't map (reverse_rz)
Initializing platform...done
Resolving host-name...done
Starting RTSP handshake...RTSP OPTIONS request failed: 404 failed: 404

hayleyhailstorm, 2018-12-16:
I also tried the steps suggested in https://github.com/irtimmer/moonlight-embedded/issues/716 to no avail

kurac338, 2018-12-16:
Following steps from this link:

https://github.com/irtimmer/moonlight-embedded/issues/716#issuecomment-445205242

should get you running. I have managed to get it working. But decided not to close the ticket since I bet more people are still struggling with it.

hayleyhailstorm, 2018-12-16:
> Following steps from this link:
> 
> [#716 (comment)](https://github.com/irtimmer/moonlight-embedded/issues/716#issuecomment-445205242)
> 
> should get you running. I have managed to get it working. But decided not to close the ticket since I bet more people are still struggling with it.

Unfortunately I tried that already and it didn't work :(

hayleyhailstorm, 2018-12-18:
I tried 3.15 of GFE with the 2.1.2 version of Moonlight, also updated/upgraded everything

knives49, 2018-12-18:
update moonlight to 2.4.7 for GFE 3.15
Monnlight 2.1.2 is for GFE 2.7 or 2.8....
compile the source of moonlight for GFE 3.16

hayleyhailstorm, 2018-12-18:
> update moonlight to 2.4.7 for GFE 3.15
> Monnlight 2.1.2 is for GFE 2.7 or 2.8....
> compile the source of moonlight for GFE 3.16

What commands do I run to get it to update to version 2.4.7?

hayleyhailstorm, 2018-12-18:
> update moonlight to 2.4.7 for GFE 3.15
> Monnlight 2.1.2 is for GFE 2.7 or 2.8....
> compile the source of moonlight for GFE 3.16

never mind... got it. however, i'm still getting the same rtsp error. though, maybe i missed the "compile the source of moonlight for GFE 3.16"?

knives49, 2018-12-18:
https://github.com/irtimmer/moonlight-embedded/issues/716#issuecomment-445205242

use this guide :)

hayleyhailstorm, 2018-12-18:
Ah yeah, but I followed those steps

hayleyhailstorm, 2019-01-15:
Still not working :(

closed 2021-07-24 by cgutman