moonlight-stream/moonlight-embedded #721: Windows keeps pressing LSHIFT after stream disconnect

Daniel2193, 2018-12-23:
GFE Version: 3.16.0.140
OS (Server): Windows 10 Home (1803)
Hardware (server): gtx 960 (4GB), i7-6700k,16GB RAM
Moonlight Embedded version: (idk) latest version via git clone (23.12.18)
Moonlight Embedded source: compiled from source (git clone) without any changes
Moonlight Embedded running on: Raspberry Pi 2B
Moonlight Embedded running on distribution: Raspbian (latest 23.12.18)

Both devices are connect via LAN (in the same network)

Verbose output `-verbose` of Moonlight Embedded: no

my command to start the stream: "moonlight stream -1080 -bitrate 5000 -app mstsc 192.168.178.53"

What is the expected result?: after stream disconnect everthing should be normal (on server)

What happens instead of that? Windows keeps LSHIFT pressed until PC (server) reboot, same happends when i reconnect with my Pi2B. I can see that because windows' internal on screen keyboard shows that. Just pressing Shift on on-screen-keyboard or my regular keyboard does nothing, CAPS is still working (if CAPS is on i write lower case letters).

Any help?

antarktikali, 2018-12-28:
I have a similar issue. If you open the on-screen keyboard on Windows, you can see that ctrl-alt-shift-q buttons remain pressed on the server after you press those buttons on the client to stop the streaming session. For everything other than shift, pressing them again will release them. However, the shift remains pressed.

If you stop the Nvidia service on the server with the following command in an admin terminal, it releases the shift key:
`net stop NvContainerLocalSystem`

Daniel2193, 2019-01-01:
> I have a similar issue. If you open the on-screen keyboard on Windows, you can see that ctrl-alt-shift-q buttons remain pressed on the server after you press those buttons on the client to stop the [...]

Seems not to work, it says that there is no such service running, even though i copy/pasted it (maybe i did sth else wrong) but you brought me to the idea that it must be a process that obviusly can be killed and after some testing i figured out that its easier to just create a `.bat` File which cointains
`taskkill -im nvstreamer.exe -f`
You just have to run that file as Administrator and than the stream disconnects and there are no keys pressed
Or you just run that File after you disconnect the stream via CRTL+ALT+SHIFT+Q (still need admin permission)
If you dont want a `.bat` File you can also just kill the `NVIDIA Streamer Server Container` Process in Taskmanager

Anyway thanks for the help @antarktikali

antarktikali, 2019-01-01:
Maybe this shouldn't be closed? That seems to be more like a workaround. To me it seems that upon exit, the client might not be shutting down properly and keeping the input on the host in a bad state.

closed 2019-01-24 by irtimmer