moonlight-stream/moonlight-embedded #722: Cant install because package doesnt update.

Doredude, 2018-12-27:
Please provide the following info.

Latest
Latest
sudo apt-get moonlight-embedded
Raspberry Pi
Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**
i dont know what this means
**What is the expected result?**
it installs
**What happens instead of that?**
it tells me that "Clearsigned file isn't valid, got 'NOSPLIT' (does the network require authentication?"
is the file not been updated or did i not do it right i followed the tutorial of this site :https://www.howtogeek.com/220969/turn-a-raspberry-pi-into-a-steam-machine-with-moonlight/

posti85, 2018-12-27:
Fetch and install the GPG key
```
wget http://archive.itimmer.nl/itimmer.gpg
sudo apt-key add itimmer.gpg
```

closed 2019-01-18 by irtimmer

Oranjoose, 2023-03-16:
> Fetch and install the GPG key
> 
> ```
> wget http://archive.itimmer.nl/itimmer.gpg
> sudo apt-key add itimmer.gpg
> ```

I'm getting the same error after adding that key. The error continues by saying "The repository 'http://archive.itimmer.nl/raspbian/moonlight buster InRelease' is no longer signed." It then refuses to update saying that updating from this repository cannot be done securely.

Is there a workaround for this?

cgutman, 2023-03-17:
These instructions are way out of date. See https://github.com/moonlight-stream/moonlight-embedded/wiki/Packages for updated instructions.

Oranjoose, 2023-03-17:
> These instructions are way out of date. See https://github.com/moonlight-stream/moonlight-embedded/wiki/Packages for updated instructions.

Running the

curl -1sLf 'https://dl.cloudsmith.io/public/moonlight-game-streaming/moonlight-embedded/setup.deb.sh' | distro=raspbian codename=buster sudo -E bash

command on Buster Raspbian results in the same clearsigned file isn't signed error.

cgutman, 2023-03-17:
You'll need to remove the old repo from your APT sources list or it will keep trying to fetch it and failing.