moonlight-stream/moonlight-embedded #724: Mouse emulation for controller like on moonlight for android

canneloni, 2019-01-04:
Please provide the following info.

**_NVidia Geforce Experience version:_**3.16.0.140
**Moonlight Embedded version:** 2.4.7
**Moonlight Embedded source:** compiled from source
**Moonlight Embedded running on:** Raspberry Pi
**Moonlight Embedded running on distribution:** Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**

Moonlight Embedded 2.4.7 (PI;SDL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.0.234...
NVIDIA GeForce GTX 1060 6GB, GFE 3.16.0.140 (gs_04_11_25137997, 7.1.411.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share//moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**

Just like on android, it would be greate to longpress start (or any other button) and have a virtual mouse.

**What happens instead of that?**

Nothing

kurac338, 2019-01-04:
+1 for this; would be awesome. Parsec also has support for this, by pressing Select + DPAD Up. A combination like so, that does not interfere with game bindings would be great.

cgutman, 2021-08-08:
Implemented by 469b2ed8668cee46dcd39c74e84d18c1f211cbc4

closed 2021-08-08 by cgutman