moonlight-stream/moonlight-embedded #725: 120 FPS

Prendaloth, 2019-01-16:
Is there any way to make moonlight embedded stream 120FPS, like moonlight Qt ?

irtimmer, 2019-01-18:
Every frequency can be specified using the option `-fps` including 120FPS. However most boards, or at least the Raspberry Pi, does not support decoding 120FPS

closed 2019-01-18 by irtimmer

Prendaloth, 2019-01-19:
I found this board for 45 $. It has 6 core CPU, way better than raspberry PI 
https://www.friendlyarm.com/index.php?route=product/product&product_id=241

CPU Model: Rockchip RK3399
Number of Cores: big.LITTLE, 64-bit Dual Core Cortex-A72 + Quad Core Cortex-A53
Frequency: Cortex-A72(up to 2.0GHz), Cortex-A53(up to 1.5GHz)
GPU: Mali-T864
4K VP9 and 4K 10bits H265/H264 60fps decoding, Dual VOP, etc

Do you think it can handle 1080p@120FPS ?