moonlight-stream/moonlight-embedded #726: Problems with Desktop streaming

doveman, 2019-01-20:
Please provide the following info.

**_NVidia Geforce Experience version:_3.15.0.186
**Moonlight Embedded version: 2.47 with latest irtimmer changes deb
**Moonlight Embedded source: repository
**Moonlight Embedded running on: Raspberry Pi
**Moonlight Embedded running on distribution Raspbian Stretch

**Verbose output `-verbose` of Moonlight Embedded:
Desktop streaming in 720P:
pi@raspberrypi:~ $ moonlight stream -verbose -app Desktop
Moonlight Embedded 2.4.7 (PI;SDL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.64...
NVIDIA GeForce GTX 1070 Ti, GFE 3.15.0.186 (gs_04_09_24784787, 7.1.409.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share/moonlight/gamecontrollerdb.txt
No mapping available for CHESEN USB Keyboard (03000000810a00000101000010010000) on /dev/input/event1
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Waiting for IDR frame
Waiting for IDR frame
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

Desktop streaming in 1080P:
pi@raspberrypi:~ $ moonlight stream -verbose -1080 -app Desktop
Moonlight Embedded 2.4.7 (PI;SDL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.64...
NVIDIA GeForce GTX 1070 Ti, GFE 3.15.0.186 (gs_04_09_24784787, 7.1.409.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share/moonlight/gamecontrollerdb.txt
No mapping available for CHESEN USB Keyboard (03000000810a00000101000010010000) on /dev/input/event1
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

Streaming without using -app Desktop:
pi@raspberrypi:~ $ moonlight stream -verbose
Moonlight Embedded 2.4.7 (PI;SDL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.64...
NVIDIA GeForce GTX 1070 Ti, GFE 3.15.0.186 (gs_04_09_24784787, 7.1.409.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share/moonlight/gamecontrollerdb.txt
No mapping available for CHESEN USB Keyboard (03000000810a00000101000010010000) on /dev/input/event1
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed
Loss Stats: Transaction failed: 11
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
I've added mstsc.exe in GFE in order to stream the desktop using -app Desktop, which should display the desktop full-screen on my TV and if I don't use -app Desktop it should launch Steam.

**What happens instead of that?**
When using -app Desktop, regardless of whether I use -1080 or let it default to 720p, the desktop streams in a box/window that doesn't fill the screen, with the Raspian desktop visible either side.

If I don't use -app then instead of launching Steam it just streams the Desktop as above. To get moonlight/GFE to launch Steam as it should I have to delete the mstsc.exe entry from GFE and then I have no way of streaming the desktop.

doveman, 2019-01-20:
I've uploaded a photo of my TV showing the problem here: https://ibb.co/sFzN1Vs

Jasperras, 2019-02-12:
I'm running into the same issue on my raspberry pi model b+ (raspbian stretch, built moonlight-embedded from latest).

I've checked if the issue also occurs with moonlight on linux, however it is correctly rendered in fullscreen there. On a side note; my host has a higher resolution than the client, and downscaling isn't automatically applied in this scenario. As such it is required to manually apply the client resolution on the host via nvidia control panel.

Thus this confirms it to be an issue with moonlight-embedded specifically.