moonlight-stream/moonlight-embedded #727: compiling for LibreELEC Generic-X86_64  errors

succo69, 2019-01-23:
ok, i'm trying to build moonlight-embedded for my LibreELEC installations; i correctly built for rpi2, it pairs but i still have to see if it actually works but really can't build for x86_64. I was trying to cmake with the toolchain file but found that it's a no way so started building within LE, as an addon, adding the package.mk files for moonlight-common-c (tried bot the irtrimmer and the moonlight-stream versions) and moonlight-embedded. i had to work something with enet to make the irtrimmer fork of moonlight-common-c to build but i successfully did it (and embedded) for the rpi2, even if i still have to try in a game. so i tried with x86_64, which would be my primary task as the main htpc is an x86_64, but it refuses to build, i can build moonlight-common-c (both -stream and -embedded versions) and enet (bot cgutman and 1.3 version) but when it comes to build libgamestream it errors like this:

[47/51] Linking C shared library libgamestream/libmoonlight-common.so.2.4.7
FAILED: libgamestream/libmoonlight-common.so.2.4.7 
: && /home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc -fPIC -march=x86-64 -m64 -mmmx -msse -msse2 -mfpmath=sse -fomit-frame-pointer -Wall -pipe -Os -Os -DNDEBUG  -march=x86-64 -m64 -Wl,--as-needed -fuse-ld=gold -shared -Wl,-soname,libmoonlight-common.so.4 -o libgamestream/libmoonlight-common.so.2.4.7 libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/AudioStream.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/ByteBuffer.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/Connection.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/ControlStream.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/FakeCallbacks.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/InputStream.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/LinkedBlockingQueue.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/Misc.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/Platform.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/PlatformSockets.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/RtpFecQueue.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/RtpReorderQueue.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/RtspConnection.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/RtspParser.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/SdpGenerator.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/SimpleStun.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/VideoDepacketizer.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/src/VideoStream.c.o libgamestream/CMakeFiles/moonlight-common.dir/__/third_party/moonlight-common-c/reedsolomon/rs.c.o  -lenet -lssl -lcrypto && :
/home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/8.2.0/../../../../x86_64-libreelec-linux-gnu/bin/ld.gold: error: /home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/libenet.a(packet.o): requires unsupported dynamic reloc 11; recompile with -fPIC
/home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/8.2.0/../../../../x86_64-libreelec-linux-gnu/bin/ld.gold: error: /home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/libenet.a(peer.o): requires unsupported dynamic reloc 11; recompile with -fPIC
/home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/8.2.0/../../../../x86_64-libreelec-linux-gnu/bin/ld.gold: error: /home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/libenet.a(protocol.o): requires unsupported dynamic reloc 11; recompile with -fPIC
/home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/lib/gcc/x86_64-libreelec-linux-gnu/8.2.0/../../../../x86_64-libreelec-linux-gnu/bin/ld.gold: error: /home/succhenrica/LibreELEC.tv/build.LibreELEC-Generic.x86_64-9.0-devel/toolchain/x86_64-libreelec-linux-gnu/sysroot/usr/lib/libenet.a(unix.o): requires unsupported dynamic reloc 11; recompile with -fPIC
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

i tried to add -fPIC to the cmakelist but it keeps not working... any help would be much appreciated

succo69, 2019-01-24:
ok, what was needed was to build enet with -fPIC, as the error itself states... once added the correctoption to the enet's project.mk and rebuilt it worked right.

closed 2019-01-24 by succo69

mittal07abhilash, 2021-02-25:
> ok, what was needed was to build enet with -fPIC, as the error itself states... once added the correctoption to the enet's project.mk and rebuilt it worked right.

Hi succo69! I want to build moonlight-embedded for my x86_64 HTPC too. Can you outline a process on how to do that. No details are required, just where and how to begin the process. I'll try to sort any errors encountered and use your previous comments as reference.