moonlight-stream/moonlight-embedded #731: Moonlight Embedded through VPN hangs on Searching Server

filthygurll, 2019-01-31:
Please provide the following info.

**_NVidia Geforce Experience version:_**
GFE 3.17.0.113

**Moonlight Embedded version:**
Moonlight Embedded 2.4.7

**Moonlight Embedded source:**
Compiled from source
https://github.com/irtimmer/moonlight-embedded.git

**Moonlight Embedded running on:**
Raspberry Pi 3 b+

**Moonlight Embedded running on distribution:** 
Retropie 4.4.5

**Verbose output `-verbose` of Moonlight Embedded:**
moonlight pair 192.168.0.4 -verbose                             Moonlight Embedded 2.4.7 (PI;SDL;X11;ALSA;PULSE;EMBEDDED)
Connect to 192.168.0.4...
NVIDIA GeForce GTX 1060 6GB, GFE 3.17.0.113 (gs_04_13_25598729, 7.1.413.0)
Please enter the following PIN on the target PC: 4058
Succesfully paired

moonlight -720 192.168.0.4 -verbose                             Moonlight Embedded 2.4.7 (PI;SDL;X11;ALSA;PULSE;EMBEDDED)
Searching for server...

**What is the expected result?**
Moonlight streams steam to the raspberry pi

**What happens instead of that?**
Moonlight hangs on "Searching for Server..."

filthygurll, 2019-01-31:
I've had this setup for over a year now but ever since the whole commit update I've been having this issue.

I have SoftEther setup on my PC in order to create a VPN wit a virtual hub which I can access with OpenVPN on the Raspberry pi, this allowed me to use moonlight away from home with pretty much any WiFi network with a strong enough connection.

cgutman, 2021-07-24:
mDNS might not work through VPNs, so just specify the IP address for streaming like you did for pairing