moonlight-stream/moonlight-embedded #733: Prob

doveman, 2019-02-08:
**_NVidia Geforce Experience version:_3.17.0.126
**Moonlight Embedded version: 2.47 with latest irtimmer changes deb
**Moonlight Embedded source: repository
**Moonlight Embedded running on: Raspberry Pi
**Moonlight Embedded running on distribution Raspbian Stretch

If I run" moonlight stream -1080 -app xx" then after I end the stream with Ctrl+Alt+Shift+Q, I can't use my mouse or keyboard on my PC. Only the left mouse button works and I can't use the Windows key and have to reboot to fix it. It seems Moonlight leaves the Alt key locked, as pressing Tab acts as if I'm pressing Alt+Tab and cycles through the open programs.

Also, the Default audio device was locked on "Nvidia Virtual Audio" and I couldn't change it to my Realtek device. After rebooting that was fixed also and the Realtek device was the only one showing under Playback devices, although the Nvidia device was still enabled in Device Manager, so I guess that it's meant to clear it out of Playback devices and reset to Realtek when the stream ends.

Also, the resolution is left on 1920x1080 after ended the stream and I have to manually set it back to 1920x1200.

This is the log I copied from the terminal after the stream terminated itself. The video had been reasonably OK but the audio was constantly suffering from quantisation errors which were quite annoying. I put up with it and played for a couple of hours before everything went uber choppy shortly before it terminated.

Received OOS audio data (expected 63006, but got 63015)
Returning RTP packet after queue overgrowth
Received OOS audio data (expected 63016, but got 63019)
Unrecoverable frame 312297: 25+0=25 received < 40 needed
Network dropped an entire frame
Unrecoverable frame 312299: 12+0=12 received < 39 needed
Unrecoverable frame 312300: 11+0=11 received < 40 needed
Unrecoverable frame 312301: 11+0=11 received < 39 needed
Unrecoverable frame 312302: 10+0=10 received < 40 needed
Unrecoverable frame 312303: 11+0=11 received < 40 needed
Unrecoverable frame 312304: 11+0=11 received < 39 needed
Unrecoverable frame 312305: 11+0=11 received < 40 needed
Unrecoverable frame 312306: 11+0=11 received < 40 needed
Unrecoverable frame 312307: 11+0=11 received < 40 needed
Unrecoverable frame 312308: 9+0=9 received < 39 needed
Unrecoverable frame 312309: 12+0=12 received < 40 needed
Unrecoverable frame 312310: 10+0=10 received < 40 needed
Unrecoverable frame 312311: 11+0=11 received < 40 needed
Unrecoverable frame 312312: 11+0=11 received < 39 needed
Unrecoverable frame 312313: 10+0=10 received < 40 needed
Unrecoverable frame 312314: 11+0=11 received < 40 needed
Unrecoverable frame 312315: 11+0=11 received < 40 needed
Unrecoverable frame 312316: 11+0=11 received < 39 needed
Unrecoverable frame 312317: 11+0=11 received < 40 needed
Unrecoverable frame 312318: 11+0=11 received < 40 needed
Unrecoverable frame 312319: 11+0=11 received < 40 needed
Returning RTP packet queued for too long
Received OOS audio data (expected 63035, but got 63037)
Unrecoverable frame 312320: 11+0=11 received < 40 needed
Returning RTP packet queued for too long
Received OOS audio data (expected 63044, but got 63089)
Returning RTP packet queued for too long
Received OOS audio data (expected 63090, but got 63103)
Returning RTP packet queued for too long
Received OOS audio data (expected 63104, but got 63107)
Returning RTP packet queued for too long
Received OOS audio data (expected 63108, but got 63121)
Returning RTP packet queued for too long
Received OOS audio data (expected 63122, but got 63191)
Network dropped an entire frame
IDR frame request sent
Waiting for IDR frame
Unrecoverable frame 312322: 11+0=11 received < 40 needed
Network dropped an entire frame
IDR frame request sent
Waiting for IDR frame
Unrecoverable frame 312324: 11+0=11 received < 40 needed
Unrecoverable frame 312325: 0+7=7 received < 40 needed
Returning RTP packet after queue overgrowth
Received OOS audio data (expected 63192, but got 63209)
Unrecoverable frame 312326: 2+0=2 received < 40 needed
Unrecoverable frame 312327: 9+0=9 received < 39 needed
Unrecoverable frame 312328: 13+0=13 received < 40 needed
Unrecoverable frame 312329: 11+0=11 received < 39 needed
Unrecoverable frame 312330: 11+0=11 received < 39 needed
Unrecoverable frame 312331: 11+0=11 received < 40 needed
[... 918 lines cut ...]
Waiting for IDR frame
Unrecoverable frame 312805: 12+0=12 received < 39 needed
IDR frame request sent
Unrecoverable frame 312806: 10+1=11 received < 40 needed
Unrecoverable frame 312807: 5+7=12 received < 39 needed
Unrecoverable frame 312808: 26+8=34 received < 40 needed
Unrecoverable frame 312809: 8+8=16 received < 39 needed
Returning RTP packet after queue overgrowth
Unrecoverable frame 312810: 25+8=33 received < 39 needed
Received OOS audio data (expected 1085, but got 1139)
Unrecoverable frame 312811: 31+0=31 received < 39 needed
Unrecoverable frame 312812: 12+0=12 received < 41 needed
Control stream received disconnect event
Loss Stats: Transaction failed: 11
Stopping input stream...done
Stopping audio stream...Unrecoverable frame 312813: 12+0=12 received < 40 needed
done
Stopping video stream...Unrecoverable frame 312814: 22+8=30 received < 40 needed
done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

hhromic, 2019-02-09:
Some weeks ago I added an option to moonlight to address this issue specifically (commit d0f5f90d).
The new option is named `-quitappafter`. It will send a quit message to the server when you end the session. The server will restore all as expected then.

Compile from  latest source and you should have it available (check with `moonlight -help` and you should see it listed). Then just invoke moonlight like this:

    moonlight stream -1080 -quitappafter -app xx

kurac338, 2019-02-13:
> Some weeks ago I added an option to moonlight to address this issue specifically (commit [d0f5f90](https://github.com/irtimmer/moonlight-embedded/commit/d0f5f90dea8c703db9e1c899746f1d49dc1cac7e)).
> The new option is named `-quitappafter`. It will send a quit message to the server when you end the session. The server will restore all as expected then.
> 
> Compile from latest source and you should have it available (check with `moonlight help` and you should see it listed). Then just invoke moonlight like this:
> 
> ```
> moonlight stream -1080 -quitappafter -app xx
> ```

Hi, is this feature supposed to quit the stream automatically after the invoked application / game exits, or only when the client sends a quit stream command?

For instance I have several games that don't support GameStream, and I had to create a .bat script for them, where I would force quit the NVStreamContainer.exe to end the session.

hhromic, 2019-02-16:
> Hi, is this feature supposed to quit the stream automatically after the invoked application / game exits, or only when the client sends a quit stream command?

No, this feature is meant to send a quit *request* to the server, then GFE gracefully closes the connection and (if the game supports it) closes the game. This is exactly like quiting Moonlight and then doing `moonlight quit` in the command line.  Is up to GFE to close the connection if you close the game.

> For instance I have several games that don't support GameStream, and I had to create a .bat script for them, where I would force quit the NVStreamContainer.exe to end the session.

The quit request should handle that, at least it does on my setup. Can you try it?

kurac338, 2019-02-16:
> No, this feature is meant to send a quit _request_ to the server, then GFE gracefully closes the connection and (if the game supports it) closes the game. This is exactly like quiting Moonlight and [...]

Yes your contribution does work, it's just that I was hoping it would trigger on game quit, and not just on 'quit request'.

If it could be configured (with an additional parameter) to also trigger on game exit, that would be awesome, since I don't really think there is a need to continue the stream after you exit the application/game. You launch a new application/game and initiate a new stream in that case. I would figure that would also be cleaner for the system, as it would allow GFE to dump any stream errors or exceptions and start fresh.

hhromic, 2019-02-17:
> If it could be configured (with an additional parameter) to also trigger on game exit, that would be awesome, since I don't really think there is a need to continue the stream after you exit the application/game.

You mean that when you quit the game from the game itself, you would like moonlight to stop the streaming (and send the quit app request too)?  For a custom-added game to GFE?

Unfortunately I'm not sure if that is actually possible because Moonlight is not aware of what you do in the games and there is no feedback from GFE regarding that.

TyanColte, 2019-05-24:
> Yes your contribution does work, it's just that I was hoping it would trigger on game quit, and not just on 'quit request'.
>
> If it could be configured (with an additional parameter) to also trigger [...]

This does work if you invoke the game via moonlight itself `moonlight -app "Fallout 4"`if you quit fallout 4 it will stop the stream you can see a list of apps currently supported by gamestream by using `moonlight list` you can also use CTRL+ALT+SHIFT+Q to stop the stream manually. I use MSTSC streaming to stream my whole desktop to play games that aren't supported native by moonlight with the `-quitappafter` switch it will quit mstsc when you ctrl+shift+alt+q to quit the stream and put the desktop back in it's original condition. I also have an emergency `fixgs` alias on my pi which will SSH into my windows 10 PC (Native SSH Server support now on CMD) and run a batch file to restart the gamestream services if the `-quitappafter` switch doesn't work properly.