moonlight-stream/moonlight-embedded #734: Master does not compile on Raspberry Pi 3 - Raspbian Stretch

McNugget6750, 2019-02-17:
I follow the instructions provided in the wiki on a fresh installation of Raspbian Stretch.

pi@raspberrypi:~ $ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 http://archive.raspberrypi.org/debian stretch InRelease
Reading package lists... Done

pi@raspberrypi:~ $ sudo apt-get install git libopus0 libexpat1 libasound2 libudev1 libavahi-client3 libcurl3 libevdev2 libenet7 libssl-dev libopus-dev libasound2-dev libudev-dev libavahi-client-dev libcurl4-openssl-dev libevdev-dev libexpat1-dev libpulse-dev uuid-dev libenet-dev cmake gcc g++ fakeroot debhelper
Reading package lists... Done
Building dependency tree
Reading state information... Done
cmake is already the newest version (3.7.2-1).
debhelper is already the newest version (10.2.5).
fakeroot is already the newest version (1.21-3.1).
g++ is already the newest version (4:6.3.0-4).
gcc is already the newest version (4:6.3.0-4).
git is already the newest version (1:2.11.0-3+deb9u4).
libavahi-client-dev is already the newest version (0.6.32-2).
libavahi-client3 is already the newest version (0.6.32-2).
libcurl3 is already the newest version (7.52.1-5+deb9u9).
libcurl4-openssl-dev is already the newest version (7.52.1-5+deb9u9).
libenet-dev is already the newest version (1.3.12+ds-2).
libenet7 is already the newest version (1.3.12+ds-2).
libevdev-dev is already the newest version (1.5.6+dfsg-1).
libevdev2 is already the newest version (1.5.6+dfsg-1).
libexpat1 is already the newest version (2.2.0-2+deb9u1).
libexpat1-dev is already the newest version (2.2.0-2+deb9u1).
libopus-dev is already the newest version (1.2~alpha2-1).
libopus0 is already the newest version (1.2~alpha2-1).
libpulse-dev is already the newest version (10.0-1+deb9u1).
libssl-dev is already the newest version (1.1.0j-1~deb9u1).
libudev-dev is already the newest version (232-25+deb9u8).
libudev1 is already the newest version (232-25+deb9u8).
uuid-dev is already the newest version (2.29.2-1+deb9u1).
libasound2 is already the newest version (1.1.3-5+rpi3).
libasound2-dev is already the newest version (1.1.3-5+rpi3).
The following package was automatically installed and is no longer required:
  alsa-base
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@raspberrypi:~ $ sudo apt-get install libraspberrypi-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libraspberrypi-dev is already the newest version (1.20181112-1).
The following package was automatically installed and is no longer required:
  alsa-base
Use 'sudo apt autoremove' to remove it.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

pi@raspberrypi:~ $ git clone https://github.com/irtimmer/moonlight-embedded.git
Cloning into 'moonlight-embedded'...
remote: Enumerating objects: 42, done.
remote: Counting objects: 100% (42/42), done.
remote: Compressing objects: 100% (30/30), done.
remote: Total 5775 (delta 18), reused 25 (delta 12), pack-reused 5733
Receiving objects: 100% (5775/5775), 10.13 MiB | 6.84 MiB/s, done.
Resolving deltas: 100% (3396/3396), done.

pi@raspberrypi:~ $ cd moonlight-embedded

pi@raspberrypi:~/moonlight-embedded $ git submodule update --init
Submodule 'common' (https://github.com/irtimmer/moonlight-common-c.git) registered for path 'third_party/moonlight-common-c'
Cloning into '/home/pi/moonlight-embedded/third_party/moonlight-common-c'...
error: no such remote ref b92e4884aabd3af865c512a20366560b337619d3
Fetched in submodule path 'third_party/moonlight-common-c', but it did not contain b92e4884aabd3af865c512a20366560b337619d3. Direct fetching of that commit failed.

pi@raspberrypi:~/moonlight-embedded $ mkdir build

pi@raspberrypi:~/moonlight-embedded $ cd build/

pi@raspberrypi:~/moonlight-embedded/build $ cmake ../
-- The C compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.1.3")
-- Found Opus: /usr/lib/arm-linux-gnueabihf/libopus.so
-- Found Broadcom: /opt/vc/include
-- Could NOT find Freescale (missing:  FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY)
-- Could NOT find Amlogic (missing:  AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY)
-- Could NOT find Rockchip (missing:  ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY)
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29")
-- Checking for module 'libevdev'
--   Found libevdev, version 1.5.6
-- Checking for module 'libudev'
--   Found libudev, version 232
-- Checking for module 'sdl2>=2.0.4'

-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'x11'
--   No package 'x11' found
-- Checking for module 'vdpau'
--   No package 'vdpau' found
-- Checking for module 'libva'
--   No package 'libva' found
-- Checking for module 'libva-x11'
--   No package 'libva-x11' found
-- Checking for module 'libpulse-simple'
--   Found libpulse-simple, version 10.0
-- Checking for module 'libcec>=4'

-- Checking for module 'egl'
--   No package 'egl' found
-- Checking for module 'glesv2'
--   No package 'glesv2' found
-- Checking for one of the modules 'uuid'
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found CURL: /usr/lib/arm-linux-gnueabihf/libcurl.so (found version "7.52.1")
-- Found OpenSSL: /usr/lib/arm-linux-gnueabihf/libssl.so;/usr/lib/arm-linux-gnueabihf/libcrypto.so (found suitable version "1.1.0j", minimum required is "1.0.2")
-- Found EXPAT: /usr/lib/arm-linux-gnueabihf/libexpat.so (found version "2.2.0")
-- Checking for module 'avahi-client'
--   Found avahi-client, version 0.6.32
-- Checking for module 'libenet'
--   Found libenet, version 1.3.12
You have called ADD_LIBRARY for library moonlight-common without any source files. This typically indicates a problem with your CMakeLists.txt file
-- Configuring done
CMake Error: CMake can not determine linker language for target: moonlight-common
CMake Error: Cannot determine link language for target "moonlight-common".
-- Generating done
-- Build files have been written to: /home/pi/moonlight-embedded/build

pi@raspberrypi:~/moonlight-embedded/build $ make
Scanning dependencies of target moonlight-common
make[2]: *** No rule to make target 'libgamestream/CMakeFiles/moonlight-common.dir/build'.  Stop.
CMakeFiles/Makefile2:163: recipe for target 'libgamestream/CMakeFiles/moonlight-common.dir/all' failed
make[1]: *** [libgamestream/CMakeFiles/moonlight-common.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

pi@raspberrypi:~/moonlight-embedded/build $ cd ..

pi@raspberrypi:~/moonlight-embedded $ make

make: *** No targets specified and no makefile found.  Stop.

pi@raspberrypi:~/moonlight-embedded $

hhromic, 2019-02-17:
The problem is that the `moonlight-common-c` third-party repository is not in sync after a recent change (2adf94398094c152296c340aec0acc2300872142). Probably it will be fixed soon, you have to be patient.

hhromic, 2019-02-17:
In the meantime, you can roll-back. After cloning the repository and before compiling:
``
git checkout d0f5f90dea8c703db9e1c899746f1d49dc1cac7e
git submodule update --recursive
``

irtimmer, 2019-02-17:
moonlight-common-c repository have been updated

closed 2019-02-17 by irtimmer

hhromic, 2019-02-17:
Can confirm it's building just fine again , Thanks.