moonlight-stream/moonlight-embedded #741: Nvidia Jetson Nano support

Helios747, 2019-04-20:
Please provide the following info.

**_NVidia Geforce Experience version:_**
NA

**Moonlight Embedded version:**
NA

**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
GitHub

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Nvidia Jetson Nano

**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Ubuntu 18.04

**Verbose output `-verbose` of Moonlight Embedded:**
NA

**What is the expected result?**
NA

**What happens instead of that?**
NA

Recently Nvidia released a $100 dev board. A SBC with a Tegra X1. It has a somewhat smaller Maxwell GPU than what shipped in the Shield TV but it is otherwise an extremely capable arm64 SBC which should be able to run Moonlight just fine. Especially if you're able to actually just use nvidia's video decoder.

When running CMake, it checks for several different vendors of GPUs, fails to match Nvidia against the list, and bombs out.

Is there any interest in supporting this board?

telamon, 2019-11-29:
Output from failed attempt to compile moonlight-embedded on a jetson nano
```
-- Could NOT find Broadcom (missing: BROADCOM_INCLUDE_DIR VCOS_LIBRARY VCHIQ_LIBRARY OPENMAXIL_LIBRARY BCM_HOST_LIBRARY) 
-- Could NOT find Freescale (missing: FREESCALE_INCLUDE_DIR KERNEL_INCLUDE_DIR VPU_LIBRARY) 
-- Could NOT find Amlogic (missing: AMLOGIC_INCLUDE_DIR AMCODEC_LIBRARY AMADEC_LIBRARY AMAVUTILS_LIBRARY) 
-- Could NOT find Rockchip (missing: ROCKCHIP_INCLUDE_DIR ROCKCHIP_LIBRARY) 
-- Checking for module 'libavcodec'
--   No package 'libavcodec' found
-- Checking for module 'libavutil'
--   No package 'libavutil' found
-- Checking for module 'libcec>=4'
--   No package 'libcec' found
CMake Error at CMakeLists.txt:166 (message):
  No video output available

-- Configuring incomplete, errors occurred!
See also "/home/flash/Downloads/moonlight-embedded/build/CMakeFiles/CMakeOutput.log".
See also "/home/flash/Downloads/moonlight-embedded/build/CMakeFiles/CMakeError.log".
```

mlaugharn, 2020-01-23:
Try
sudo apt install libp8-platform-dev libcec-dev libva-dev libva2 libavcodec57 libavcodec-dev

mlaugharn, 2020-01-24:
btw it worked when I used the sdl platform instead of the default 'auto'. There may be other platforms that may even work better but it's the only one I've tried so far. It did 60 fps no problem.

`moonlight stream -platform sdl -windowed <ip address>`

0yueyunfei0, 2020-03-11:
Any hardware video decoder support news ? :)

KiralyCraft, 2021-05-04:
This is now supported by Moonlight-Qt