moonlight-stream/moonlight-embedded #742: rock pi 4 support

gillesboisson, 2019-04-27:
Please provide the following info.

Compiling last version in rock pi 4 the last version (commit 7b46b4b2ea4e51bc758a02d18cf80313ba442161)

I had compiler error :
**moonlight-embedded/src/video/rk.c**
error: 'DRM_FORMAT_NV12_10' undeclared (first use in this function)
           ret = drmModeAddFB2(fd, frm_width, frm_height, fmt == MPP_FMT_YUV420SP ? DRM_FORMAT_NV12:DRM_FORMAT_NV12_10, handles, pitches, offsets, &frame_to_drm[i].fb_id, 0);

I compiled on debian stretch, based on wiki compilation tutorial.

It seems to be a compatibility issue with the GPU [Mali T86](https://github.com/rockchip-linux/libmali/issues/26). 

As this version compiled fine and work perfectly on a pi3, but just in case you can sort out the issue. 

Thanks

iMartyn, 2019-11-18:
I can confirm that the same is true on a debian Buster Pinebook Pro.  This would be a very nice thing to fix if you can.

jgonyea, 2020-03-15:
I'd love to see moonlight working on the Pinebook Pro as well.  If it matters, I'm running Ubuntu Bionic on mine.

gponick, 2020-05-13:
I was able to get this to build for me on the RK3399 based NanoPC-T4 (running FriendlyDesktop 18.04) by essentially disabling 10 bit support here.

Editing this line: https://github.com/irtimmer/moonlight-embedded/blob/0c6f98e544720e7c39fe4c5e2e64628edb03aa19/src/video/rk.c#L200

I changed 
```          ret = drmModeAddFB2(fd, frm_width, frm_height, fmt == MPP_FMT_YUV420SP ? DRM_FORMAT_NV12:DRM_FORMAT_NV12_10, handles, pitches, offsets, &frame_to_drm[i].fb_id, 0);```
to 
```          ret = drmModeAddFB2(fd, frm_width, frm_height, DRM_FORMAT_NV12_10, handles, pitches, offsets, &frame_to_drm[i].fb_id, 0);```

Everything works but I suspect HDR will not work as this references 10 bit color. I'm going to poke at defining the format rather than removing support. I've joined the discord for moonlight and the `embed-dev` chat and will continue to update this and there.

gponick, 2020-05-14:
So poking at this more, I was able to actually just define the right definition for `DRM_FORMAT_NV12_10` - but again having no HDR support, I can't guarantee that part works. 

For reference, this is how I defined the `DRM_FORMAT_NV12_10` define, at https://github.com/irtimmer/moonlight-embedded/blob/0c6f98e544720e7c39fe4c5e2e64628edb03aa19/src/video/rk.c#6
```
// DRM_FORMAT_NV12_10
#ifndef DRM_FORMAT_NV12_10
#define DRM_FORMAT_NV12_10     fourcc_code('N', 'A', '1', '2') /* 2x2 subsampled Cr:Cb plane */
#endif // DRM_FORMAT_NV12_10
```

commit 6f15b257 (2021-08-08): Add DRM_FORMAT_NV12_10 definition for RK backend

Fixes #742
closed 2021-08-08 by cgutman