moonlight-stream/moonlight-embedded #745: Segmentation fault when

HelenFlamer, 2019-05-01:
**NVidia Geforce Experience version:** 3.18.0.102 (gs_04_13_26051692, 7.1.413.0)
**Moonlight Embedded version:** Moonlight Embedded 2.4.8 (PI;ALSA;EMBEDDED)
**Moonlight Embedded source:** deb package from http://archive.itimmer.nl/raspbian/moonlight stretch main
**Moonlight Embedded running on:** Raspberry Pi 3B
**Moonlight Embedded running on distribution:** Retropie

**Verbose output `-verbose` of Moonlight Embedded:**
pi@retropie:~/RetroPie/roms/moonlight $ moonlight stream -bitrate 100 -app "Assassin" --verbose --input=/dev/input/event4
Moonlight Embedded 2.4.8 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.10.125...
NVIDIA GeForce GTX 1060 6GB, GFE 3.18.0.102 (gs_04_13_26051692, 7.1.413.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (lefttrigger:+a3)
Can't map (righttrigger:-a3)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Can't map (dpdown:+a1)
Can't map (dpleft:-a0)
Can't map (dpright:+a0)
Can't map (dpup:-a1)
Add input /dev/input/event4...
Detected Wireless Controller (050000004c050000c405000000810000) on /dev/input/event4 as Wireless Controller
Stream 1280 x 720, 60 fps, 100 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Segmentation fault

**What is the expected result?:** Streaming starts properly

**What happens instead of that?**: 
Game successfully starts on PC, so no issues on this side.
moonlight dies on the Raspberry without any additional information or trace :(

Tried to play with the settings like bitrates, resolutions, etc, but got no success.

Please advise if I miss anything

endlessborland, 2019-05-01:
I got the same problem running Raspbian Stretch.

irtimmer, 2019-05-01:
Fixed with version 2.4.9

closed 2019-05-01 by irtimmer