moonlight-stream/moonlight-embedded #747: PS3 controller left trigger stays pressed

Pakoach, 2019-05-18:
Please provide the following info.

**_NVidia Geforce Experience version:_**
3.18.0.102

**Moonlight Embedded version:**
From master branch 763bcb76d6e69e939

**Moonlight Embedded source:**
compiled from source
https://github.com/irtimmer/moonlight-embedded.git

**Moonlight Embedded running on:** 
Raspberry Pi

**Moonlight Embedded running on distribution:** 
OSMC

**Verbose output `-verbose` of Moonlight Embedded:**
osmc@osmc:~$ moonlight stream -app "Assassin's Creed: Origins" -verbose
Moonlight Embedded 2.4.9 (PI;SDL;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.76...
NVIDIA GeForce GTX 770, GFE 3.18.0.102 (gs_04_13_26051692, 7.1.413.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share//moonlight/gamecontrollerdb.txt
Detected Sony Computer Entertainment Wireless Controller (050000004c0500006802000000810000) on /dev/input/event1 as Sony Computer Entertainment Wireless Controller
Detected Sony Computer Entertainment Wireless Controller Motion Sensors (050000004c0500006802000000810000) on /dev/input/event0 as Sony Computer Entertainment Wireless Controller
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Error: cannot keep up
ENet disconnect event pending                       <------- After closing the game
Server notified termination reason: 0x0100
Control stream received disconnect event
Failed to send ENet control packet
Loss Stats: Transaction failed: 0
Failed to send ENet control packet
Input: sendInputPacketOnControlStream() failed: -1
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
When left trigger is not pressed, it should not be pressed in the game.

What happens instead of that?**
Left trigger is always pressed in the game.
I think it was working before but I was using the older version without SDL 2 game controller DB.

The PS3 controller I use is 050000004c0500006802000000810000.
I tried to troubleshoot the issue and it seems that updating the gamecontrollerdb.txt can fix the issue. But I am not sure if it comes only from there or not.

I also noticed something unexpected. When creating the mapping manually with the command:
moonlight map -input /dev/input/event1

I get the following line for my 2 PS3 controllers (they have the same ID):
`050000004c0500006802000000810000,Sony Computer Entertainment Wireless Controller,a:b0,b:b1,x:b3,y:b2,start:b9,guide:b10,back:b8,leftstick:b11,rightstick:b12,leftshoulder:b4,rightshoulder:b5,dpup:b13,dpleft:b15,dpdown:b14,dpright:b16,leftx:a0,lefty:a1,rightx:a3,righty:a4,lefttrigger:a2,righttrigger:a5,lefttrigger:b6,righttrigger:b7,platform:Linux`

It seems that the left trigger is repeated twice with 2 different values as well as right trigger (lefttrigger:a2,righttrigger:a5,lefttrigger:b6,righttrigger:b7). I don't know if it is an error or not during the mapping command. 
In gamecontrollerdb.txt, only the first value appears (lefttrigger:a2,righttrigger:a5). I tried to use both values at the same time (like the map command suggest) and it did not fix it.

But if I use the second values instead (lefttrigger:b6,righttrigger:b7), everything works fine and lefttrigger does not stay pressed.
Do you think updating the DB is enough? Should I post a request to update SDL 2 database?
Or could it be due to the way moonlight-embedded uses the game controller inputs?

guibera, 2019-06-11:
Thank you, you saved me!

Pakoach, 2019-06-22:
I am happy it helped someone.
I posted an issue also on gabomdq/SDL_GameControllerDB to update the gamecontrollerdb.txt file as there were no answer from @irtimmer yet.
Unfortunately I don't have any PS4 controller but if the issue is fixed for PS3 controllers you should post something for PS4 as well!

Pakoach, 2020-01-13:
After posting on SDL bugzilla, they are wondering if this could be a moonlight bug. If anyone from moonlight-embedded wants to join the conversation, here is the post:
https://bugzilla.libsdl.org/show_bug.cgi?id=4686

railei, 2020-10-01:
I had the same issue with my PS3 controller. For my setup it seems to be caused by the "Motion Sensors" device. When I start the client using "-input /dev/input/event1" (i.e. disable usage of motion sensors device), left trigger behaves as expected...

cgutman, 2021-07-26:
Motion sensor bug was fixed in v2.5.0

closed 2021-07-26 by cgutman