moonlight-stream/moonlight-embedded #750: Not working on new raspbian version buster

TyanColte, 2019-06-26:
Not working on new raspbian version. Buster. Looks like some dependancies are missing. Can we get an update please?

posti85, 2019-06-27:
Did you try compiling the project from the source? (https://github.com/irtimmer/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC)

JorisVanEijden, 2019-07-02:
Use libcurl4 in stead of libcurl3 and it'll compile fine.

TyanColte, 2019-07-18:
@posti85 i just reinstalled the dependencies via the jessie repo after changing it in my sources.list and then updated it from the original repo. It's working now. At least until he adds the buster repo. @JorisVanEijden i don't compile from source cause i have too many issues when i do.

TyanColte, 2019-12-01:
@irtimmer any update on this? I'd really like to be able to install this without having to downgrade my dependencies or having to manually compile.

TyanColte, 2019-12-02:
Ok found the buster repo in the wiki. I had to purge it and reinstall it in order to download the buster build. But now I'm getting a "Problem with SSL certificate" when I try to pair with my computer. @irtimmer any suggestions?

XionXinX, 2019-12-02:
I am also having this issue on a fresh install.

kurac338, 2020-10-20:
Any update on this? I just got on the error train, with the latest OSMC 2020-10 update. They've only now started using Debian Buster...

posti85, 2020-10-20:
Hi @kurac338, I still didn't upgrade my OSMC installation because I have many things running on my Rpi and I still don't want to deal with problems related to that upgrade (these days I don't hace much time).

Anyway, did you try with a fresh install?

kurac338, 2020-10-21:
> Hi @kurac338, I still didn't upgrade my OSMC installation because I have many things running on my Rpi and I still don't want to deal with problems related to that upgrade (these days I don't hace much time).
> 
> Anyway, did you try with a fresh install?

Yes, I switched my sources list from "stretch" to "buster" main releases. I ran apt-get purge moonlight-embedded and reinstalled it. This fixed unsatisfied dependency "urllib3" errors, however now Luna won't pair with the host I had saved.

I am guessing this is similar to @TyanColte 's comment. Specifically I am getting a "InsecureRequestWarning", saying that the HTTPS call was used without a verified certificate.

Meaning that I can never get the host to pair successfully in the "buster" version of the package.

posti85, 2020-10-21:
@kurac338 I was talking about a fresh install of the whole OSMC system. Make a backup of the SD Cars and give a try, it could be worth

kurac338, 2020-10-21:
@posti85 No, I kept my existing install, since I did not experience any issues in migrating from Debian Jessie to Debian Stretch, so I did not expect any issues in this upgrade.

I am not sure that a clean install would solve this issue. My guess is that it is caused by a legacy Python dependency issue, since Luna uses urllib2, and OSMC/Kodi wraps those calls into the urllib3/4 module.

Unfortunately since Luna's developer stopped pushing updates, it was only a matter of time when something like this would happen. A shame really. And without any proper substitution to replace the Luna addon, we are stuck with the moonlight-embedded CLI (extremely inconvenient for a home media center setup without a keyboard).

I saw that @cgutman was working on moonlight-qt for RPI, but it requires a X11 interface (Raspbian), and that does not work for us OSMC (Debian) users. I was really hoping that the QT version could be packaged as a Kodi addon and that would be a great replacement for the Luna addon.

cgutman, 2020-10-21:
The Moonlight Qt client does not require X11. It runs just fine from the console.

https://github.com/moonlight-stream/moonlight-docs/wiki/Installing-Moonlight-Qt-on-Raspberry-Pi-4

kurac338, 2020-10-21:
@cgutman Nice, I did not see that documentation reference. I have tried it on OSMC, but:

`osmc@osmc:~$ moonlight-qt       
00:00:00 - Qt Info: Unable to detect Wayland or X11, so EGLFS will be used by default. Set QT_QPA_PLATFORM to override this.
00:00:00 - Qt Warning: Unable to find a KMSDRM display device!
00:00:00 - Qt Warning: On the Raspberry Pi, you must enable the 'fake KMS' driver in raspi-config to use Moonlight outside of the GUI environment.
00:00:00 - SDL Info (0): Compiled with SDL 2.0.12
00:00:00 - SDL Info (0): Running with SDL 2.0.12
00:00:00 - Qt Fatal: Could not find DRM device!
Aborted`

I could not find a way to enable 'fake KMS' on OSMC, since they don't allow the user to access raspi-config. Googling for a workaround did not yield any results. Maybe I am querying the wrong keywords, as I'm not familiar with it's purpose.

@irtimmer On #775 it was reported that this issue was resolved in the latest release 2.4.11, but I am still getting the error:
`Failed to pair to server: Problem with the local SSL certificate`

kurac338, 2020-10-22:
Ok so my headache saga has been resolved. Turns out in the migration from "Stretch" to "Buster" versions of OSMC, there were a couple of issue with moonlight-embedded.
1) I needed to update the "/etc/apt/sources.list.d/moonlight.list" to use the "buster" repository
2) I needed to purge and (re)install the moonlight-embedded package
3) I needed to clear (remove) the generated cache folder ~/.cache/moonlight, as it was causing SSL certificate validation issues

cgutman, 2021-07-24:
Fixed

closed 2021-07-24 by cgutman