moonlight-stream/moonlight-embedded #751: No video on Raspberry Pi 4

JorisVanEijden, 2019-07-02:
Please provide the following info.

**_NVidia Geforce Experience version:_**
3.19.0.107
**Moonlight Embedded version:**
2.4.10 (master commit 039eabca1e9fa8addafde826e4e6f4f675b3bca9)
**Moonlight Embedded source:**
compiled from source
**Moonlight Embedded running on:** 
Raspberry Pi 4
**Moonlight Embedded running on distribution:** 
Raspbian GNU/Linux 10 (buster)

**Verbose output `-verbose` of Moonlight Embedded:**
`Moonlight Embedded 2.4.10 (PI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.1.119...
NVIDIA GeForce GTX 980, GFE 3.19.0.107 (gs_04_15_26420198, 7.1.415.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share//moonlight/gamecontrollerdb.txt
Stream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done`

**What is the expected result?**
Game is visible on Raspberry Pi connected display.

**What happens instead of that?**
Game is launched on PC and visible there. Audio plays on the Raspberry Pi display, Raspberry Pi controllers control the game.

So everything seems to work fine and I cannot find any error logs. But the Raspberry Pi display doesn't show the video. I just keep seeing the console I launch moonlight from.
Probably due to the new GPU of the rpi4.

posti85, 2019-07-03:
Although I still don't own a Raspberry Pi 4, I've do some research. It seems [moonlight-embedded/src/video/pi.c](https://github.com/irtimmer/moonlight-embedded/blob/master/src/video/pi.c) implementation is based on [raspberrypi/firmware/opt/vc/src/hello_pi/hello_video/video.c](https://github.com/raspberrypi/firmware/blob/master/opt/vc/src/hello_pi/hello_video/video.c). Did you success on running the hello_pi/hello_video sample? If you didn't, maybe the way the video is decoded in this device has changed, and the pi.c implementation should be revised (if it's the only file involved in this issue, I really don't know it).

JorisVanEijden, 2019-07-03:
Well, I guess it was something on my OS. After a fresh install the compile went well and I now have video as well as audio and controls.
My apologies for the false report.

On a full official Rasbian install all I needed was:
```
sudo apt-get install cmake libasound2-dev libopus-dev libevdev libevdev-dev libudev-dev libcurl4-openssl-dev libssl-dev libavahi-client-dev libenet-dev libsdl2-dev uuid-dev
```
and then to work around an ssl issue edit `/etc/ssl/openssl.cnf` and on the last  line change the `DEFAULT@SECLEVEL` from 2 to 0.

closed 2019-07-03 by JorisVanEijden

zartinn, 2019-07-17:
@JorisVanEijden I have exactly the same problem. 
I would rather avoid reinstalling my OS because I've updated it already to the latest drivers and made 4k working.
I am using exactly the same versions as Joris and used the following tutorial: https://github.com/irtimmer/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC

But I had to install some newer lib because the older one was not available for buster. Anyway the compilation went smooth. Have you adjusted your sources.list at your reinstall?