moonlight-stream/moonlight-embedded #754: Unable to pair due to SSL certificate error

aSempruch, 2019-07-05:
**_NVidia Geforce Experience version:_** 3.19.0.107
**Moonlight Embedded version:** 2.4.10
**Moonlight Embedded source:** compiled from source
**Moonlight Embedded running on:** Raspberry Pi 4 Model B
**Moonlight Embedded running on distribution:** Raspbian Buster

**Verbose output `-verbose` of Moonlight Embedded:**
```
$ moonlight -verbose pair 192.168.7.104
Moonlight Embedded 2.4.10 (PI;ALSA;PULSE;EMBEDDED)
Connect to 192.168.7.104...
NVIDIA GeForce GTX 1080 Ti, GFE 3.19.0.107 (gs_04_15_26420198, 7.1.415.0)
Please enter the following PIN on the target PC: 5830
Failed to pair to server: Problem with the local SSL certificate
```
```
$ sudo moonlight -verbose pair 192.168.7.104
Moonlight Embedded 2.4.10 (PI;ALSA;PULSE;EMBEDDED)
Connect to 192.168.7.104...
NVIDIA GeForce GTX 1080 Ti, GFE 3.19.0.107 (gs_04_15_26420198, 7.1.415.0)
Please enter the following PIN on the target PC: 0470
Failed to pair to server: Problem with the local SSL certificate
```
(first time execution included a line `Generating certificate...`)

I am unable to pair with my PC due to SSL certificate errors. I am not sure what the cause might be as this is a fresh raspbian installation with no modifications made, other than the installation of moonlight and libraries mentioned in the wiki build guide necessary for compilation.

I've also tried pairing with the moonlight-qt client on my laptop and did not encounter any issues, which leads me to believe that this is a problem on the client side.

Any help would be appreciated 😄

aSempruch, 2019-07-05:
This [issue]( https://github.com/irtimmer/moonlight-embedded/issues/751) talks about the same problem and the fix is to edit `/etc/ssl/openssl.cnf` and on the last line change the DEFAULT@SECLEVEL from 2 to 0. This worked for me, however I am not sure if this is a fix or more of a workaround. I am not knowledgeable enough about what this option does, hopefully someone who is can say if leaving this option at 0 is a potential security risk.

closed 2019-07-06 by irtimmer

luca992, 2020-01-03:
I just built from master and I am running into this issue still.  The 2 to 0 work around is also the only way I can get pairing to work

griffin4747, 2026-09-30:
Remove all entries from Sunshine. Re-add worked for me.

PR #755 (2019-07-06): Use a SHA-256 signature for the client certificate to fix TLS on Buster
**Description**
SHA-1 signed certificates have been phased out by CA and SSL client implementations. On Debian Buster, the default OpenSSL policy rejects SHA-1 certificates and prevents Moonlight from working.

**Purpose**
Stop generating new SHA-1 client certificates to fix pairing and streaming on Debian/Raspbian Buster with the default OpenSSL config.

Fixes #754