moonlight-stream/moonlight-embedded #756: 2 mouse cursors showing. Raspberry PI 4 - Buster and Windows

Philosophise, 2019-07-12:
Hi,

When I stream my windows desktop I get both the cursor from windows and also the cursor from the raspberry pi. The windows desktop streams perfectly.

Interestingly the cursor from the raspberry pi seems to execute whatever it clicks on on the raspberry pi desktop (although I cannot see the raspberry pi desktop).

Thank you 

GeForce Experience 3.19.0.107
Moonlight Raspberry PI 4 (running on Buster) v 2.4.10

pi@raspberrypi:~ $ moonlight stream -verbose -app mstsc -1080 -fps 60
Moonlight Embedded 2.4.10 (PI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.0.27...
NVIDIA GeForce GTX 1080 Ti, GFE 3.19.0.107 (gs_04_15_26420198, 7.1.415.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share/moonlight/gamecontrollerdb.txt
No mapping available for MOSART Semi. 2.4G Wireless Mouse (030000002a0600000641000010010000) on /dev/input/event1
Stream 1920 x 1080, 60 fps, 20000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed: -1
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

aandaluz, 2019-08-22:
I have faced a similar issue, where mouse clicking events on the remote desktop would also be forwarded to the local host.  So double clikcing on the remote would ocassionally open a folder in the host if I had a opened XFCE  file manager window.

After going through the moonlight-embbeded commits, I've found that commit [212eac6 ](https://github.com/irtimmer/moonlight-embedded/commit/212eac6359a5bfae262ea2d6ccecd477bec38275) ("Only grab input devices if they are keyboards")  introduced this problem, since the previous commit [[d7ac865]](https://github.com/irtimmer/moonlight-embedded/commit/d7ac8654e58f5a5f309b8043f82a69b92062863d)  fixed the issue for me .  Reverting the changes introduced by [212eac6]  on master also works .

I'm currently using Geforce experience 3.17 with Driver 419.67 with Maxwell GPUs in the remote host.

closed 2019-09-13 by irtimmer

aandaluz, 2019-09-16:
Just writting to confirm that the PR fixed the issue and mouse/keyboard events no longer affect local screen when interacting with remote hosts.
Still I'm facing a different random issue which sometimes leaves the raspberry host frozen after disconnect. I'll file a separate bug If I can reproduce it.

commit 243ef8ae (2019-09-13): Also grab mouse input devices (fixes #756)