moonlight-stream/moonlight-embedded #757: Resolution scaled to Host screen for games manually added to GFE

maretodoric, 2019-09-07:
Please provide the following info.

NVidia Geforce Experience version: 3.20.0.118
Moonlight Embedded version: Moonlight Embedded 2.4.10
Moonlight Embedded source: repository
Moonlight Embedded running on: Raspberry Pi 3B+
Moonlight Embedded running on distribution: Raspbian

Verbose output `-verbose` of Moonlight Embedded:
`Moonlight Embedded 2.4.10 (PI;ALSA;EMBEDDED)
Connect to 192.168.0.100...
NVIDIA GeForce GTX 1060 6GB, GFE 3.20.0.118 (gs_04_15_26867986, 7.1.415.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Add input /dev/input/by-id/usb-Logitech_Wireless_Gamepad_F710_8DFA9604-event-joystick...
Detected Logitech Gamepad F710 (030000006d0400001fc2000005030000) on /dev/input/by-id/usb-Logitech_Wireless_Gamepad_F710_8DFA9604-event-joystick as Logitech F710 Gamepad (XInput)
Add input /dev/input/by-id/usb-ShanWan_PC_PS3_Android-event-joystick...
Detected ShanWan PC/PS3/Android (03000000632500007505000010010000) on /dev/input/by-id/usb-ShanWan_PC_PS3_Android-event-joystick as SHANWAN PS3/PC Gamepad
^[OPStream 1280 x 720, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...ENet wait interrupted
Control stream connection failed: -1
done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
`

**What is the expected result?**
When game is launched, it should be displayed fullscreen on TV that is connected to Raspberry Pi...

**What happens instead of that?**
...But it's somehow scaled to screen size of Host PC (21:9) even though resolution is changed.

This would happen if game is added manually to GeForce Experience through "Settings/SHIELD/ADD".
However, if i launch steam, it will Launch it in Big Picture Mode with normal resolution and as well as any game launched within Steam. It will maintain normal res.
However, some games, for example Mortal Kombat X, I've added it as Non Steam shortcut inside steam, when i try to launch it from moonlight from within steam big picture, it will open up Steam Store. This works fine when not streaming.

maretodoric, 2019-09-07:
OK, so this would happen because the resolution will be bound to desktop resolution. It doesn't seem to matter if the resolution is different in game. Can i make moonlight change resolution on host PC when streaming starts ?
I've created a batch that would change resolution, launch game then revert the resolution back to original one, then converted that batch to exe, but it's not an elegant solution.

cTatu, 2020-01-17:
@maretodoric I have the same issue. Could you please share that batch script for changing the resolution?

maretodoric, 2020-01-18:
> @maretodoric I have the same issue. Could you please share that batch script for changing the resolution?

It's much easier this way, this is how I'm using it.
```

@ECHO OFF
SET APP="Steam.exe"
SET APPPATH="C:\Program Files (x86)\Steam\Steam.exe"

tasklist /FI "IMAGENAME eq %APP%" 2>NUL | find /I /N %APP%>NUL
if "%ERRORLEVEL%"=="0" (
	echo Steam je pokrenut, menjam rezoluciju ...
	QRes /x:1280 /y:720
	start %APPPATH% "steam://open/bigpicture"
	goto LOOP1
	:LOOP1
	tasklist /FI "IMAGENAME eq %APP%" 2>NUL | find /I /N %APP%>NUL
	if "%ERRORLEVEL%"=="0" (
		TIMEOUT /T 10 > NUL
		goto LOOP1
	) else (
		goto END
	)
	
) else (
	QRes /x:1280 /y:720
	%APPPATH% -bigpicture
	goto END
)

:END
QRes /x:2560 /y:1080
```

What this would do, it will check if the steam is already running, and if it is, it will change the resolution and just put it to big picture mode. From there you can launch any games you like (even if you add a non steam game). At steam exit it will revert back the resolution.

You need QRes program, download it and place it in C:/Windows/system32 (that program will allow you to change resolution from batch script).
Add new item to nVidia Experience and point to batch file, and don't forget to edit your resolution preferences and APP and APPPATH variables.

cTatu, 2020-01-21:
Thank you but that doesn't helped me in any way. Turns out this client it's not exactly as the moonlight-qt client and doesn't held well different aspect rations source-destination images. I commented it in [this issue](https://github.com/irtimmer/moonlight-embedded/issues/364#issuecomment-576926404) hoping they could fix it more easily.