moonlight-stream/moonlight-embedded #761: [Feature] Option to disable input

tadly, 2019-09-17:
Would it be possible to add an option to ignore all input?

I use moonlight **just** as a video feed while using peripherals directly connected to the client and would like to ensure not interference between the two input *methods*.

yukatan, 2019-10-03:
I'm do it this just using steam. You can go to gamepad config and hide any control you don't want to use

hhromic, 2019-10-05:
Hi, I just implemented such an option now in Moonlight embedded. See the referenced PR.
Testing and feedback is appreciated. Thanks!

The new command-line option is `-viewonly` and it can also be stored in config using `viewonly = true`.

tadly, 2019-10-07:
@yukatan for what I'm doing, steam is waaaaay to bloated but nice to know you can do that

@hhromic awesome, I have a super busy schedule but hope I can get around testing it sometime soon.
Thanks a lot for working on this :)

hhromic, 2019-10-07:
@tadly ah dang :/
Whenever I get access to an X11-based computer I'll try to implement the feature there.
Sorry the current implementation doesn't apply to you at the moment.

PR #764 (2019-10-08): Add option for disabling all input (view-only mode) in streaming session
**Description**

This PR adds a new option to enable a "view-only" mode, that is, for Moonlight embedded to not send any input back to the host computer and just act as a remote viewer.

NOTE: at the moment this is only implemented for embedded and SDL platforms, not for X11 due to the way the latter's input system is implemented. I may implement for X11 in the future.

I tested this on a Raspberry Pi device only (embedded case), but it should work on SDL-based devices as well. Testing on these is appreciated.

I also cle [... cut]
closed 2019-10-08 by irtimmer