moonlight-stream/moonlight-embedded #762: Sutters on pi 3B and 3B+

yukatan, 2019-09-27:
I'm totally bewildered by something that happens to me using moonlight. I've been trying different configurations and the result is always the same. I'm using moonlight on a rasperry pi 3B and I have intermittent sttutering all the time. Sometimes it goes well gives a couple of sutters then it goes well and then it goes back to having sutters for a while making the game experience unpleasant.

In the logs everything seems to be right, no drop frames, no error messages.

If I try the same configuration from an Android Samsung Galaxy S9 client with Samsung DEX the experience is perfect 1080@60. no sutters. The server is the same and all devices, both the raspberry and the phone are connected by ethernet.

To rule out a problem with the raspberry itself and OSMC which is the system I have installed, I have done another test with a raspberry pi 3B+ and a clean installation of Arch Linux. The result is exactly the same.

My questions are : Is it a performance problem of the raspberry 3?  Would it be better to use raspberry pi 4 B? Is there anyone who has a raspberry 4 that can tell if it improves the experience with this more powerful device? Is it a known limitation in raspberry pi 3 and I'm wasting my time trying to fix it?

I can put the logs if it is needed but there is everything right......

Mirarchi, 2019-09-30:
I have a Raspberry PI 4 and sometimes there Is a microstutter. I dont know why but It Is really annoying... I can't understand of It Is a decoder problem or maybe something in the acceleration video.

yukatan, 2019-10-01:
I got a raspberry pi 4 for testing and with a clean instalation of arch linux it has exactly the same stutters than the pi 3B and 3B+ so is not a performance problem.

Something is not working in moonlight-embedded.

The android client works as expected

himuura, 2019-10-22:
Mates, what OS's are you using? My safe bet would be it's a OS issue (if it's buster based) or a dependency package that got "updated" and now it's not working correctly. I would assume that you have either a Samsung Class 10 microSD or an SSD/USB device to compensate the wear of the microSD card? Would somewhat improve performance overall but i think it's not the main reason for the stutters (moonlight is network-hungry not disk-eater). Other issues might arrise with an insuficient Power Supply Unit or Thermal related issues. If i muster the strength to reinstall raspbian buster today, i will do a trial run ( i have the sniffles :(   ). But feel free to take a look at my guesses. will keep you posted!

himuura, 2019-10-24:
No issues whatsoever with RPI 3. Tested for around 15min though, didn't have much time. Compiled it from scratch, no .deb or repo was used (just follow the instructions and replace libcurl3 requisite with libcurl4). RPI 3B with Samsung Evo Class 10 microSD, 3 copper heatsinks (SoC, RAM and USB/Ethernet controller), opened case on the sides, small fan attached to the top acrylic part blowing air into the SoC heatsink. Both the Pi and the server machine were cable connected. Hope this helps you solve your problem!Cheers!

yukatan, 2019-11-27:
@himuura I have the same config.. rapbian buster lite in microsd samsung evo class 10. Heatsinks and small fan. Cat6 cable for host and raspberry pi 3 ( tested as well connected directly to host without switch or router ). Moonlight compiled from source. Network 1Gbs (tested host with 100Mbs) always microstutters... power supply is 5V 3Amp but no official one so I bought the official to check.. we will see but I don't see any power alert. Host is  I7 3770k 16gb ram and gtx1080

yukatan, 2019-12-01:
After test with the official power supply, same stutters. I tested multiple raspbian versions. Multiple raspberry pi versions (3B,3B+,4B). Compiled from source or not. I did a clean instalation of windows 10. I did changes in all posible ethernet options in the host. I try a external usb3 to ethernet in the host... testing the network with iperf3 no packet loss and no jitter. Android client works perfect  with "nerver drop frames"  in samsumg galaxy s9 and xiami my box s (with a little more latency) so I think is not the network or the host. Raspberry pi works ok for some minutes, then some microstutters, then ok for a while... and this loop all the time.

yukatan, 2020-05-14:
Fixed with a new switch with flow control enabled. it was a network issue.

closed 2020-05-14 by yukatan