moonlight-stream/moonlight-embedded #765: RTSP OPTIONS request failed: -1

DarthMudder, 2019-10-07:
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...Failed to parse RTSP response
RTSP OPTIONS request failed: -1
failed: -1
Cleaning up platform...done

I am using the addon Luna on OSMC to launch Moonlight.
I have reinstalled the nVidea driver and moonligt.
Even Moonlight started via PuTTy but the same error occurs. 

How can I fix this bug ?

hhromic, 2019-10-09:
Are you using the BitDefender antivirus?
This is a known recent bug with that antivirus and is not Moonlight's fault. The antivirus HTTP proxy implementation is faulty in recent versions. Solution is to disable the antivirus or wait for it to be fixed. I understand BItDefender is working on a fix already.

DarthMudder, 2019-10-09:
Thank you for the quick response and help.
I am using Bitdefender Antivirus and have deactivated it. Without success.
Now I uninstalled Bitdefender and it works now.
What can I do to make it work with Bitdefender ?
Can I filter it via Bitdefender or create a rule so it doesn't interfere ?

hhromic, 2019-10-09:
In the Moonlight Discord server, a user shared the following email from a BitDefender representative regarding the issue. You can try the suggested workaround. I don't use that Antivirus so can't help you further. You can also join the Moonlight Discord, there is a channel dedicated to this issue.

---

Thank you for your patience throughout this entire process and please accept my apologies for the frustration that we might have caused.

I have looked into the case and it seems that the situation is caused by an issue which has already been acknowledged by our development team. We are currently working on a fix which will be done by means of automatic update and won't require any additional action on your end.

We will get back to you with an update regarding the fix as soon as we have one.

In the meantime, as a workaround you can apply the following modification:
1. Restart your device in safe mode as guided below:
http://www.bitdefender.com/support/how-do-i-restart-in-safe-mode-1022.html

2. Navigate to the file path below:

C:\Program Files\Bitdefender\Bitdefender Security

3. Rename the "httpproxy.dll" file to "httpproxy.dllx" without the ""

4. Restart back in normal mode.

Thank you once again and have a nice day ahead!

Best regards,
Andrei Moagar
Senior Technical Support Engineer

DarthMudder, 2019-10-09:
Wow, thanks for the help.
Now everything works.

closed 2019-10-09 by DarthMudder