moonlight-stream/moonlight-embedded #766: Some keyboard keys aren't working

meister-kruemel, 2019-10-20:
NVidia Geforce Experience version: 3.20.0.118
Moonlight Embedded version: Moonlight Embedded 2.4.10
Moonlight Embedded source: Source
Moonlight Embedded running on: Raspberry Pi 3B+
Moonlight Embedded running on distribution: Raspbian Stretch Lite

Verbose output -verbose of Moonlight Embedded:

```
Moonlight Embedded 2.4.10 (PI;ALSA;EMBEDDED)
Connect to 192.168.100.23...
NVIDIA GeForce GTX 1060 6GB, GFE 3.20.0.118 (gs_04_15_26867986, 7.1.415.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/share/moonlight/gamecontrollerdb.txt
Stream 1680 x 1050, 60 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
ENet disconnect event pending
Server notified termination reason: 0x9304
Disconnect event timeout expired
Stopping input stream...done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done
```

What is the expected result?
All keyboard keys should work.

What happens instead of that?
The following keys using a german QWERTZ keyboard are not working correctly within Moonlight session (marked yellow):

[image]

1. < >: no function.
2. Windows-key: no function.
3. AltGr: acts like pressing left Alt key, so some special characters (the blue ones) are not useable.

They work perfectly fine in Raspbian, as well as the rest of the keys in both Raspbian and Moonlight. 

Any ideas?

Thx!

rindeastwood, 2022-11-04:
Same here but just for the <> key. Works perfectly fine within raspbian.

NVidia Geforce Experience version: 1.4.1.11
Moonlight Embedded version: Moonlight Embedded 2.5.3
Moonlight Embedded running on: Raspberry Pi 4
Moonlight Embedded running on distribution: Raspbian 10 (buster)

acidbubble, 2023-03-21:
Another case of everything working wonderfully except for  `<>`. Feels like the those don't get forwarded to the remote at all

Moonlight Embedded version: Moonlight Embedded 2.5.3
Moonlight Embedded running on: Raspberry Pi 3B+
Moonlight Embedded running on distribution: Raspberry Pi OS (Buster)

robberval, 2025-06-22:
I can confirm this problem still persists.
Also tested with PT-BR ABNT2 keyboard. In this mapping, the "Context menu key" doesn't work, nor Numpad's period, nor the "/ and ?"
"Barrier KVM Switch" sends the keys corretly. Perhaps you could take a look at Barrier's code to see if you can get some inspiration for a solution.