moonlight-stream/moonlight-embedded #767: missing libcurl3 but only 4 is available

leee8675, 2019-11-14:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.20.1.57
**Moonlight Embedded version:** not sure, have not been able to install it
**Moonlight Embedded source:** deb http://archive.itimmer.nl/raspbian/moonlight stretch main
**Moonlight Embedded running on:** _Raspberry Pi
**Moonlight Embedded running on distribution:** Raspbian_

**Verbose output `-verbose` of Moonlight Embedded:** N/A

**What is the expected result?** Installing Moonlight

**What happens instead of that?** I cannot install this program without libcurl3 unless there is a new version of this that I can get.

[image]
[image]

LupusVII, 2019-11-20:
Hi ! 
I add some troubles with this message but I manage to make my way by following this thread : https://github.com/irtimmer/moonlight-embedded/issues/716
Remove libcurl3 from the list of dependancies to fetch since it provokes errors.

irtimmer, 2019-11-30:
Duplicate of #750

closed 2019-11-30 by irtimmer