moonlight-stream/moonlight-embedded #775: problem with the local SSL certificate

Lightace-dk, 2020-01-04:
Please provide the following info.

**_NVidia Geforce Experience version: 3.20.2.34, also tried with 3.19.0.107
**Moonlight Embedded version:  2.4.10 
**Moonlight Embedded source: package from http://archive.itimmer.nl/raspbian/moonlight buster main
**Moonlight Embedded running on:  raspberry pi 4B, 4G
**Moonlight Embedded running on distribution: raspbian buster , clean install,  updated 4th january 2020, Linux raspberrypi 4.19.75-v7l+ #1270 
downloaded image 2019-09-26-raspbian-buster-full and wrote it with belenaEtcher
**Verbose output `-verbose` of Moonlight Embedded:**
moonlight -verbose
Moonlight Embedded 2.4.10 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 172.20.0.147...
NVIDIA GeForce GTX 1060 3GB, GFE 3.20.2.34 (gs_04_15_27341342, 7.1.415.0)
Please enter the following PIN on the target PC: 6805
Failed to pair to server: Problem with the local SSL certificate

**What is the expected result?**
That moonligt would pair to my computer. I already have my shield tablet K1 paired and working. 
**What happens instead of that?**
Failed to pair to server: Problem with the local SSL certificate.
I have tried to use the -unsupported option, 
with sudo and without, 
have tried to use the -keydir  /home/pi/.cache/moonlight 
have tried to delete the certificates in the dir, and new ones get created. But all gives the same result.
Connected via Wifi 5GHz

yaoyi098, 2020-01-08:
Same problem. My device is Pi 3

vilkoz, 2020-01-09:
I had the same problem on PI 3B on Raspbian buster
The error occurs after the call of function `http_requst` at `libgamestream/client.c:600` with the following parameters:
```
url: https://192.168.0.113:47984/pair?uniqueid=063e17b1c4e6e714&uuid=0d5e8ffb-3edb-4870-bbd6-e8389bb8f398&devicename=roth&updateState=1&phrase=pairchallenge
data->memory: 0x1bbaa60
data->size: 0
```

Curl version:
```
pi@raspberrypi:/tmp/moonlight-embedded/build $ curl --version
curl 7.64.0 (arm-unknown-linux-gnueabihf) libcurl/7.64.0 OpenSSL/1.1.1d zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3
Release-Date: 2019-02-06
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz TLS-SRP HTTP2 UnixSockets HTTPS-proxy PSL 
pi@raspberrypi:/tmp/moonlight-embedded/build $ apt list --installed | grep curl
curl/stable,now 7.64.0-4 armhf [installed]
libcurl3-gnutls/stable,now 7.64.0-4 armhf [installed,automatic]
libcurl4-openssl-dev/stable,now 7.64.0-4 armhf [installed]
libcurl4/stable,now 7.64.0-4 armhf [installed]
```

**Fixed** the problem after these steps:
```
sudo apt update
sudo apt upgrade
rm -rf /home/pi/.cache/moonlight/
```

Lightace-dk, 2020-01-11:
Hmmm tried to update and remove the .../.cache/moonlight folder.
**curl --version:**
curl 7.64.0 (arm-unknown-linux-gnueabihf) libcurl/7.64.0 OpenSSL/1.1.1d zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3
Release-Date: 2019-02-06
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz TLS-SRP HTTP2 UnixSockets HTTPS-proxy PSL 
**curl --version**
curl 7.64.0 (arm-unknown-linux-gnueabihf) libcurl/7.64.0 OpenSSL/1.1.1d zlib/1.2.11 libidn2/2.0.5 libpsl/0.20.2 (+libidn2/2.0.5) libssh2/1.8.0 nghttp2/1.36.0 librtmp/2.3
Release-Date: 2019-02-06
Protocols: dict file ftp ftps gopher http https imap imaps ldap ldaps pop3 pop3s rtmp rtsp scp sftp smb smbs smtp smtps telnet tftp 
Features: AsynchDNS IDN IPv6 Largefile GSS-API Kerberos SPNEGO NTLM NTLM_WB SSL libz TLS-SRP HTTP2 UnixSockets HTTPS-proxy PSL
**installed curl packages:**
curl/stable,now 7.64.0-4 armhf [installed]
libcurl3-gnutls/stable,now 7.64.0-4 armhf [installed,automatic]
libcurl4/stable,now 7.64.0-4 armhf [installed,automatic]
I see that I am missing **libcurl4-openssl-dev/stable,now 7.64.0-4 armhf [installed]** when i compare the output to yours. Can this cause the issue?

It will generate new files 
pi@raspberrypi:~ $ ls -la /home/pi/.cache/moonlight/
total 24
drwxr-xr-x 2 pi pi 4096 Jan 11 14:09 .
drwxr-xr-x 8 pi pi 4096 Jan 11 14:09 ..
-rw-r--r-- 1 pi pi 2430 Jan 11 14:09 client.p12
-rw-r--r-- 1 pi pi 1082 Jan 11 14:09 client.pem
-rw-r--r-- 1 pi pi 1704 Jan 11 14:09 key.pem
-rw-r--r-- 1 pi pi   16 Jan 11 14:09 uniqueid.dat

cgutman, 2020-01-11:
I confirmed this locally on a fresh Raspbian install.

@irtimmer The 2.4.10 packages for Buster are effectively broken out of the box because they are missing https://github.com/irtimmer/moonlight-embedded/commit/519a14a42786bafd267a215f671d4d73ccf7b13e, so Buster's default OpenSSL policy won't allow our client certificate to be used.

I think all we need here is to tag and release 2.4.11 packages. That should take care of these problems.

Pacos, 2020-01-15:
Same issue here on a new Raspbian install. 

I tried this workaround :

> sudo apt update
> sudo apt upgrade
> rm -rf /home/pi/.cache/moonlight/

But it didn't work, I still got the same error message : Failed to pair to server: Problem with the local SSL certificate

Steam link works correctly on the host, as I was able to connect a laptop to it through the Windows release of Moonlight.

vilkoz, 2020-01-15:
@Lightace-dk @Pacos try to compile moonlight-embedded from sources, then delete keys generated by moonlight-embedded from the package repository.
cgutman mentioned the root cause of the issue in his comment.

Pacos, 2020-01-17:
@vilkoz : I gave it a try. I couldn't find rbp-userland-osmc and rbp-userland-dev-osmc in the raspbian repositories, so I quickly gave up :(

Lightace-dk, 2020-01-18:
@vilkoz : Thank you, I successfully installed from source, and it solved the problem. I have now paired my devices.
@Pacos : Give it another try, you do not need the osmc packages for raspian. I follwed this guide: [https://github.com/irtimmer/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC](url) litterally copy pasting all the commands. 
In section 3 I read it as either OSMC or raspian, and you only have to do raspian part in section 3.

Pacos, 2020-02-18:
@Lightace-dk : I gave it another try. It was indeed way easier with all the commands already prepped (on my previous attempt I tried to manually add all dependencies). It now works :) Thanks for your help !

dpo007, 2020-03-08:
I'm having this problem as well. latest version sof everything.  Tried wiping Moonlight packages and cache, and then building from source as laid out here.  No problems during build,  but still complains about the local SSL certificate when I try to pair. :/

cgutman, 2020-03-10:
The fix for this has been released in v2.4.11.

Pacos, 2020-03-11:
Great news !

closed 2021-07-24 by cgutman