moonlight-stream/moonlight-embedded #776: no audio + no video stream on client (RPi2 osmc)  (host

klen-ghost, 2020-01-07:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.11.0.73 (also on 3.20)
     nvidia display driver 411.63 
**Moonlight Embedded version:** latest  (how do i check to be sure?)
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._ 
               deb http://archive.itimmer.nl/raspbian/moonlight stretch main

**Moonlight Embedded running on:** 
    Raspberry Pi2 B

**Moonlight Embedded running on distribution:** 
osmc (2019-11-1 kernel Linux 5.19.55-6)
**Verbose output `-verbose` of Moonlight Embedded:**
osmc@osmc:~$ moonlight pair 192.168.2.10
Connect to 192.168.2.10...
Please enter the following PIN on the target PC: 1470
Succesfully paired
osmc@osmc:~$ moonlight stream 192.168.2.10
Connect to 192.168.2.10...
No mapping available for Gaming Keyboard G110 Consumer Control (030000006d0400002ac2000010010000) on /dev/input/event3
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Returning RTP packet queued for too long
Received OOS audio data (expected 11158, but got 11159)

**What is the expected result?**
functional moonlight with audio & video
**What happens instead of that?**
blackscreens, host pc does start steam big picture

yukatan, 2020-04-17:
I don't know if this can help you, but run "systemctl stop mediacenter" before moonlight

ferdef, 2021-09-29:
Same happens to me as well. When Steam starts, sound is played through the RPi, but it cuts and no video

cgutman, 2021-09-30:
I’m going to guess that’s because it’s using Full KMS mode which doesn’t support OMX/MMAL overlays.

Can you post your /boot/config.txt?