moonlight-stream/moonlight-embedded #777: Rotated output -- random timing

DMCShep, 2020-01-14:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.20.2.34 (latest)
**Moonlight Embedded version:** Latest master as of 2020-01-13
**Moonlight Embedded source:** compiled from source
**Moonlight Embedded running on:** _Raspberry Pi 4B 1GB
**Moonlight Embedded running on distribution:** Raspbian Buster (latest)

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**
The image displays in the correct orientation

**What happens instead of that?**
It displays in the correct orientation, but also sometimes does this:
[image]

DMCShep, 2020-01-16:
Are there debug logs somewhere I can provide on top of the `-verbose` option? It still seems pretty bare-bones, and doesn't clue me into much of what is happening behind the scenes. If a log exists I will happily grab that!

Here is what I have attempted to fix the issue since I seem to be the only one with it and the unofficial Discord was also baffled as to a cause:

__Self-Diagnostic steps:__
- Fixed overscan via a TV setting. I tried adjusting `/boot/config.txt` and even though I got all the settings mostly right and even managed to shove them to absolute extremes like the below pic with `200` on all four sides (proving a point of if that had any reasonable effect mostly), overscan was still occuring in the stream itself which was fully unaffected:
[image]

- `raspi-config` , tried both `enable 4k60` and `disable 4k60`. Disabling 4k60 seemed to fix it for a longer duration however the screen did eventually rotate on its own with zero input on the PC, and rotated back also.

__Things to try now:__
- Use my 4k monitor instead of my TV to see if this is related to my TV
- Use a spare 1080p monitor to see if this is an upscaling issue

Issue as it stands is still unresolved. Determining if it is a code bug or a problem with my individual setup should prove insightful that said.

BeaudanBrown, 2020-01-24:
I am also getting this, incredibly annoying
Raspberry pi 4 B
moonlight-embedded from apt-get
Latest buster

DMCShep, 2020-01-24:
There is a suspicion of it being related to the use of an HDMI switch and Pi 4 B compatibility with them. I have no idea why that would cause flipping, that still makes zero sense to me but at least there is a working theory.

Both of us chatted on the Discord and we are borh using HDMI switches so should have an update tonight. If it is in fact an HDMI switch problem, I would like to see in the troubleshooting guide a mention of this issue and a potential fix since it is such an easy one. Fingers crossed!

BeaudanBrown, 2020-01-24:
The issue is confirmed still happening for me even with the switch removed 😭

DMCShep, 2020-01-29:
Some combination of:

1) Removing HDMI Switch
2) Changing HDMI port from one furthest away from charging port
3) Removing microHDMI to HDMI adapter and using a wire with the correct ends (without adapter)

Fixed my flipping issue. Not sure which one was the culprit or why but I tested for several hours and zero issues during this timeframe. It's the longest I've been able to do it, so I'll go ahead and call this one good.

I will wait on confirmation from @BeaudanBrown to see if these fixed his issue before closing this ticket.