moonlight-stream/moonlight-embedded #778: failed to pair to server:Problem with the local SSL certificate

14072398tom, 2020-01-14:
Please provide the following info.

**_NVidia Geforce Experience version:3.20.2.34
**Moonlight Embedded version:latest?
**Moonlight Embedded source:github
**Moonlight Embedded running on:** _Raspberry Pi 4B+
**Moonlight Embedded running on distribution:Raspbian buster

**Verbose output `-verbose` of Moonlight Embedded:**

**What is the expected result?**run

**What happens instead of that?**failed to pair to server:Problem with the local SSL certificate

I tried as the issue#555 suggest to install the libssl1.0.0 but it just won't work,the installion was success,but after enter the pair code,the "failed to pair to server:Problem with the local SSL certificate"keeps shown.

need help!

DMCShep, 2020-01-14:
#775 maybe? Check this comment in particular, worked for me on a fresh install:
https://github.com/irtimmer/moonlight-embedded/issues/775#issuecomment-572784509

14072398tom, 2020-01-15:
problem solved，I fellowed the instruction below：
https://github.com/irtimmer/moonlight-embedded/issues/754
change“2”to“0”is very useful！

closed 2020-01-15 by 14072398tom