moonlight-stream/moonlight-embedded #781: Handshaking problem.

Kewlpatate, 2020-04-15:
NVidia Geforce Experience version: 3.20.3.63
Moonlight Embedded version: 2.4.10
Moonlight Embedded source: package
Moonlight Embedded running on: Raspberry Pi
Moonlight Embedded running on distribution: Raspbian Buster

Verbose output -verbose of Moonlight Embedded:

Moonlight Embedded 2.4.10 (PI;ALSA;PULSE;EMBEDDED)
Connect to 192.168.2.20...
NVIDIA GeForce RTX 2060 SUPER, GFE 3.20.3.63 (gs_04_22_28196077, 7.1.422.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share//moonlight/gamecontrollerdb.txt
No mapping available for SIGMACHIP USB Keyboard System Control (030000004f1c00000200000010010000) on /dev/input/event3
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP message too long
RTSP DESCRIBE request failed: -1
failed: -1
Cleaning up platform...done

I recently changed GPU (with updated drivers) and since I can't pass the handshake part on my raspberry. Only a blackscreen and I have to alt-shift-ctrl-Q to exit.

I tried reinstalling (with my limited knowledge).
No firewall or antivirus "ON".
It still work with my android tablet.

cgutman, 2020-04-15:
You need to be on Moonlight Embedded 2.4.11 to stream from GeForce Experience 3.20.3. Try updating.

Kewlpatate, 2020-04-15:
Well,
How do you update ? (I'm pretty noob with this)
Only thing I can do is getting 2.4.10 using the method in the wiki.

Thank you.

codinghipster, 2020-05-05:
Might be worth updating the release notes to call this out. 2.4.10 calls out support for 3.19 but it's not explicit that 2.4.11 is required for 3.20 support.

gadgethm, 2020-05-18:
I have this issue as well, running on the version of moonlight-embedded from the AUR.  It looks like it has been a while since the AUR was updated as well.

Couldn't figure out how to get the newer version of moonlight-embedded to work on Arch, so I downgraded GeForce Experience to 3.19.0.107.  With the older versions, everything seems to play nice.

closed 2021-07-24 by cgutman