moonlight-stream/moonlight-embedded #789: Unable to pair/connect with a desktop PC - Can't connect to server 192.168.0.5

belegdol, 2020-05-24:
Please provide the following info.

**_NVidia Geforce Experience version:_** 3.20.3.63
**Moonlight Embedded version:** 2.4.11-1buster1
**Moonlight Embedded source:** repository
**Moonlight Embedded running on:** Raspberry Pi 1B
**Moonlight Embedded running on distribution:** Raspbian

**Verbose output `-verbose` of Moonlight Embedded:**
```
$ moonlight --debug pair
Moonlight Embedded 2.4.10 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.0.5...
Request https://192.168.0.5:47984/serverinfo?uniqueid=9935fb6b4e705f19&uuid=9679db78-583a-4325-b255-54b0df233da1
Request http://192.168.0.5:47989/serverinfo?uniqueid=9935fb6b4e705f19&uuid=5b7e5568-94ed-4f20-8bbd-a872d2015e37
Can't connect to server 192.168.0.5
```
**What is the expected result?**
connection is established

**What happens instead of that?**
nothing

I have enabled streaming in GFE:
[image]

DraXus, 2020-08-28:
Same issue here. Using this setup:

- NVidia Geforce Experience version: 3.20.4.14
- Moonlight Embedded version: 2.4.11-1buster1
- Moonlight Embedded source: repository
- Moonlight Embedded running on: Raspberry Pi 4
- Moonlight Embedded running on distribution: Raspbian

belegdol, 2020-09-24:
I fixed my problem by reinstalling GFE.

closed 2020-09-24 by belegdol