moonlight-stream/moonlight-embedded #795: RPi4 moonlight-embedded crashes after ~10min

foxofox, 2020-07-11:
Hi. I am running version 2.4.10 on a RPi4 buster version.

When I stream over LAN (Ethernet) my RPi at some point (~10min) crashes to black screen (no sound, nothing as if shut down). The PC I am streaming from continues running without any issues. 

What to do? And how can I see the log so I can give u some more info?

Thanks guys!

irtimmer, 2020-07-12:
Run moonlight with the option `-verbose` and copy the output after moonlight crashes.

foxofox, 2020-07-12:
How can I access the output if the whole RPi freezes?

irtimmer, 2020-07-13:
You can save the output to a file by appending `&> outputfile` to the Moonlight command and check the file after the Pi crashed. However since the whole Pi freezes I doubt this is a bug in Moonlight but rather a power or overheating issue.

foxofox, 2020-07-14:
Thanks for the info. I will see what I can do and post.
In terms of Pi freezing (power, heat): Why would that happen? What can I do
here? What would you do?

mylovemyway, 2020-07-14:
What might be the case, is the screen saver kicking in. I experienced similar issues, which I resolved by disabling the screen saver. I used "xscreensaver" to achieve this.

foxofox, 2020-07-15:
Wow! That seemed to be the issue. Thank you so much.
I have tested 30 mins yesterday and so far everything looks good. Will test
further on the weekend and report back!

I would never have thought of that :D

manaboy, 2020-11-08:
> What might be the case, is the screen saver kicking in. I experienced similar issues, which I resolved by disabling the screen saver. I used "xscreensaver" to achieve this.

Screensaver oh the pc end or pi?

foxofox, 2020-11-08:
Pi.