moonlight-stream/moonlight-embedded #800: PS4 controller is detected as two XBOX controllers in Steam Big Picture

LucasStromberg, 2020-10-11:
Please provide the following info.

**_NVidia Geforce Experience version:_**

**Moonlight Embedded version:**
2.4.10
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Raspberry Pi
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Raspbian (Retropie image)

**Verbose output `-verbose` of Moonlight Embedded:**
moonlight stream -1080
Searching for server...
Connect to 192.168.0.195...
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Error: cannot keep up
Error: cannot keep up
Returning RTP packet queued for too long
Received OOS audio data (expected 246, but got 248)
Unrecoverable frame 678: 8+0=8 received < 9 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Returning RTP packet queued for too long
Received OOS audio data (expected 4610, but got 4612)
Server notified termination reason: 0x0100
Control stream received disconnect event
Failed to send ENet control packet
Loss Stats: Transaction failed: 0
Stopping input stream...Failed to send ENet control packet
Input: sendInputPacketOnControlStream() failed: -1
done
Stopping audio stream...done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result?**
I'm expecting my PS4 controller to be recognized normally in Steam Big Picture.

**What happens instead of that?**
The controller is being recognized as two XBOX controllers.

I have tried turning off and on the "support for..." settings with no change.
Opening Big Picture on my PC directly detects no controllers.

LucasStromberg, 2020-10-12:
I updated the verbose output since I did it wrong the last time. I see now that three separate PS4 entries are detected. In Big Picture three separate controllers are actually detected sometimes.

Any way to disable the motion and touchpad entries?

kurac338, 2020-10-19:
@LucasStromberg I had a similar issue way back when I was initially trying to set up Moonlight using different DS4 (v1 and v2) controllers on a Raspberry Pi 3 running OSMC.

More specifically I had really terrible input lag caused by the gimmicky touchpad and motion sensors.

You can permanently disable those features / inputs using these contents:

`# Disable DS4 motion controls

# NOTES:
# cat /proc/bus/input/devices
# udevadm test <Sysfs path from cat>

SUBSYSTEM=="input", ATTRS{name}=="*Wireless Controller Motion Sensors", ENV{ID_INPUT}="", ENV{ID_INPUT_JOYSTICK}="", ENV{ID_INPUT_ACCELEROMETER}="", RUN+="/bin/rm %E{DEVNAME}"
SUBSYSTEM=="input", ATTRS{name}=="*Wireless Controller Touchpad", ENV{ID_INPUT}="", ENV{ID_INPUT_JOYSTICK}="", ENV{ID_INPUT_TOUCHPAD}="", RUN+="/bin/rm %E{DEVNAME}"`

in your "/etc/udev/rules.d/51-disable-DS4-excess-inputs.rules" file. They will apply when you reboot the system and connect your controller.

LucasStromberg, 2020-11-09:
Hi again. With some help from the Discord channel I realized that I could simply pass the desired controller (/dev/input/event2) with the _-input_ flag. This works quite well for now!

GregDMeyer, 2020-11-23:
I ran into this issue today too---took the udev path and it works great. (seems like the other way works too, though). @LucasStromberg how did you determine which /dev/input/* device was the one you wanted? trial and error?

GregDMeyer, 2020-11-23:
(btw I think the DS4 appears as not just 2 but in fact 3 xbox controllers, if you touch the touchpad!)

LucasStromberg, 2020-11-23:
@GregDMeyer Yes, by trial and error. I'm sure that you could write a script that automatically finds which one to use, but for me the input numbers haven't changed, so hardcoding input 2 worked for me at least.

cgutman, 2021-07-26:
Fixed in v2.5.0

closed 2021-07-26 by cgutman