moonlight-stream/moonlight-embedded #801: In-game video stream freezes but audio is working, for certain games

LucasStromberg, 2020-10-12:
Please provide the following info.

**_NVidia Geforce Experience version:_**
3.20.4.14

**Moonlight Embedded version:**
2.4.10

**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
deb http://archive.itimmer.nl/raspbian/moonlight wheezy main

**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Raspberry Pi

**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Raspbian (Retropie)

**Verbose output `-verbose` of Moonlight Embedded:**
moonlight stream -1080 -verbose
Moonlight Embedded 2.4.10 (PI;MMAL;SDL;X11;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.0.195...
NVIDIA GeForce GTX 1070, GFE 3.20.4.14 (gs_04_24_28591776, 7.1.424.0)
Platform Raspberry Pi (Broadcom)
Loading mappingfile /usr/local/share//moonlight/gamecontrollerdb.txt
Detected Sony Interactive Entertainment Wireless Controller (030000004c050000cc09000011810000) on /dev/input/event2 as PS4 Controller
Detected Sony Interactive Entertainment Wireless Controller Touchpad (030000004c050000cc09000011810000) on /dev/input/event0 as PS4 Controller
Detected Sony Interactive Entertainment Wireless Controller Motion Sensors (030000004c050000cc09000011810000) on /dev/input/event1 as PS4 Controller
Stream 1920 x 1080, 30 fps, 10000 kbps
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...done
Initializing control stream...done
Initializing video stream...done
Initializing audio stream...done
Initializing input stream...done
Starting control stream...done
Starting video stream...done
Starting audio stream...done
Starting input stream...done
Error: cannot keep up
Error: cannot keep up
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Returning RTP packet queued for too long
Received OOS audio data (expected 1968, but got 1970)
Returning RTP packet queued for too long
Received OOS audio data (expected 2124, but got 2126)
Returning RTP packet queued for too long
Received OOS audio data (expected 8602, but got 8606)
Returning RTP packet queued for too long
Received OOS audio data (expected 14352, but got 14356)
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Returning RTP packet queued for too long
Received OOS audio data (expected 22762, but got 22764)
Unrecoverable frame 3620: 13+5=18 received < 23 needed
Network dropped an entire frame
Waiting for IDR frame
IDR frame request sent
Returning RTP packet queued for too long
Received OOS audio data (expected 24244, but got 24246)
^CStopping input stream...done
Stopping audio stream...ENet wait interrupted
Control stream connection failed: -1
done
Stopping video stream...done
Stopping control stream...done
Cleaning up input stream...done
Cleaning up audio stream...done
Cleaning up video stream...done
Cleaning up control stream...done
Cleaning up platform...done

**What is the expected result and what happens instead of that?**
When starting certain games (e.g. Borderlands 3, Red Dead Redemption 2) the screen turns to black and freezes, while the audio works fine. On my PC everything is normal.
By Alt+Tab:ing out and in of the game makes the stream work again for Borderlands 3, but for RDR2 the stream only updates to the latest frame from the PC, and then freezes. E.g. going out and in of RDR2 makes the title screen visible instead of the black screen, but it's still frozen.
Steam Big Picture works completely fine, and some games (e.g. Goat Simulator) doesn't freeze.

manaboy, 2020-11-07:
Did this issue ever get fixed?

cgutman, 2020-11-07:
This isn't a Moonlight bug. It's an issue with GeForce Experience's game capture. Moonlight just displays the video it receives.

Since video comes back to life when another app is in the foreground, it's definitely a server side issue. You can try to adjust the in-game settings (Vulkan vs DirectX and full-screen vs borderless windowed) to workaround the issue.

manaboy, 2020-11-07:
Right but I've experienced this issue's with Moonlight-NX for the Nintendo switch, audio would stop other times video would stop and other times the enter app would freeze and would let you even exit to game list. Using the android version on my tablet or phone don't suffer this issue, also the pc client also don't suffer this issue. any idea what's wrong?

manaboy, 2020-11-07:
Moonlight-NX is based on embedded Moonlight.

cgutman, 2020-11-08:
Is there a GitHub page for Moonlight-NX?

manaboy, 2020-11-08:
https://github.com/rock88/moonlight-nx

cgutman, 2020-11-09:
I would advise filing an issue on that GitHub page, since it seems slightly different than the issue that was described here.

LucasStromberg, 2020-11-09:
Hi, sorry for responding so late. I did not find an explicit fix, but I found a weird workaround by swapping my main pc monitor to my second one. Doing this and the stream no longer freezes for me.

I don't know why this could be, since both monitors are connected to my graphics card.
However the second monitor (the one working) is DVI->HDMI, while the other one (not working) is DP. Could this make a difference? 

I'm not sure if this will work forever, but it's a temporary one working for me at least.

manaboy, 2020-11-12:
I've just had a similar odd issue, so recently I sorted out the hdmi mess behind my old Samsung ue65hu7500 4k TV, I basically disconnected all the hdmi cable from the TV and reconnected them neatly. Whenever I tried to load any game, from Moonlight android, Moonlight-NX (nintendo switch) Moonlight psVita, Moonlight windows, it would load to a black screen, like completely black screen, I thought my screen went to sleep as it looked like there was no backlit light, I found that when I turn on my TV and change the TV source to pc via hdmi Moonlight gives a "connection terminated" error and the TV displays my desktop, now if I try to load any game with Moonlight whilst my TV is on, it loads fine and works like normal, if I exit the game and switch off my TV, Moonlight would go back to blank screen mode on any re launch of any game.  Anyway after spending some time, I remembered your post and tried something.

The workaround was to connect the pc hdmi cable to my TV hdmi port 2 that says "HDMI2/DVI" now with the TV off Moonlight works, even after a pc reboot. How strange.

manaboy, 2020-11-12:
For Moonlight to work, your pc must be connected via HDMI to your TV/Monitors HDMI port that says "HDMI/DVI, or if your connecting to a monitor you have to connect via DVI, or if your monitor has a port that says HDMI/DVI connect it to that. 

Is this a issue with Nvidia experience or Moonlight core?

Leophir, 2021-01-16:
Set your FPS to 30 whenever you face this issue. This is an issue with Nvidia experience.