moonlight-stream/moonlight-embedded #803: Stream command fails with RTSP Error.

Fredrum, 2020-10-26:
**NVidia Geforce Experience version:**
3.18.0.102

**Moonlight Embedded version:**
2.4.11

**Moonlight Embedded source: repository/included in distribution/compiled from source/...**
https://github.com/irtimmer/moonlight-embedded.git

**Moonlight Embedded running on: Raspberry Pi/Cubox-i/Hummingboard/Other linux device/...**
RPi4 - Buster

**Moonlight Embedded running on distribution: Arch Linux/Raspbian/OpenELEC/...**
Raspbian

**Verbose output -verbose of Moonlight Embedded:**
Moonlight Embedded 2.4.10-master-5b6639c

What is the expected result?
Having done the pairing with PC successfully and then launching with 'stream' I expected it to start doing that.

What happens instead of that?
It started well but then I got:
```
Starting RTSP handshake.....RTSP OPTIONS request failed.
Failed: 552
```

The the Raspberry froze completely and I had to reboot.

I am trying to run over LAN with both RPi and PC attached to the same network switch.
The PC has its firewall disable for testing.

Any ideas of what causes this to fail?

Cheers
Fred

der-ranjit, 2020-11-07:
Getting the same error, althought the pi doesn't freeze.

Fredrum, 2020-11-07:
I never continued with this but i read that windows 8.1 does not support rtsp and thats the win version i was running.

M1reyu, 2020-12-09:
I get the exact same error message. GFE states that gamestreaming should be possible from windows 7 upwards. Is it really only the operating system that causes the problem?

Fresh install. Tried per package install and also per cmake.
RPi3B+ Buster
GFE 3.20.5.70 Win7
Moonlight Embedded 2.4.10-master-5b6639c

```
Starting RTSP handshake...RTSP OPTIONS request failed: 552
failed: 552
```

Then the raspberry freezes. Sometimes I also get additionally:
```
Cleaning up platform...done
Error: cannot keep up
```

Fredrum, 2020-12-09:
Hi I don't have any more info about this as I haven't kept trying to get it to work.
But I believe there's a 'Qt Moonlight' version that also supports the RPi that might be more actively maintained?
Might be worth checking?

irtimmer, 2020-12-09:
@Fredrum suggesting using alternative software isn't very useful in solving bugs

Fredrum, 2020-12-09:
@irtimmer  You're welcome to suggest something yourself! Complaining about others isn't useful in solving bugs

IceCloud76, 2021-07-01:
Well I'm running Windows 10 and I'm having the same issue.

Raspberry Pi 3 with Retropie
Windows 10 PC and a GTX 960

voggur, 2021-07-20:
Moonlight Embedded 2.4.10
I have the same problem.

tiesselune, 2021-09-22:
Moonlight Embedded 2.5.1, Raspberry 4, GFE 3.23.0.74 and same problem as well... Before the 552 error code, I get a 
```
RTSP port : 48010 (RTSP URL parsing failed) 
```
I don't know if it is relevant. 

The RTSP handling part is in the [Moonlight common C](https://github.com/moonlight-stream/moonlight-common-c) dependency, so I guess opening an issue there might be a good idea ?