moonlight-stream/moonlight-embedded #810: RTSP message too long - moonlight-commom-c fork

peetie2k, 2021-04-11:
Hi,

after a recent GFE Update to 3.22.0.28 I'm facing RTSP Connection Errors. I'm using your library via https://github.com/clarkemw/moonlight-embedded-launcher.
`
Searching for server...
Connect to 192.168.178.166...
Initializing platform...done
Resolving host name...done
Starting RTSP handshake...RTSP message too long
RTSP DESCRIBE request failed: -1
failed: -1
Cleaning up platform...done
`

Seems like the base of your fork did some adjustments to conform to GFE 3.22 RTSP (beside some more stuff):
https://github.com/moonlight-stream/moonlight-common-c/commit/9b194cc700fb598591ba042d45d4060288d16c4a

Would be great if you could adjust accordingly. Many Thanks :)

joaopccosta, 2021-04-12:
+1 I too am having this very same issue.

DonSpaggiari, 2021-04-13:
Very same issue. It all started with the latest GeForce Experience update. Rolling back to a previous version restores functionality as a temporary workaround. I can provide more information of requested, but the original post describes the exact same problem I’m having.

AdenGames, 2021-04-17:
+1 same issue

Vegz78, 2021-04-17:
Same issue here as well after the latest GFE update.

Also unable to cmake the updated repository according to the wiki instructions on RPi OS buster(seg fault for testCCcompiler.c).

A new release is greatly appreciated! Used almost every day by my son on his RPi 3.

Thanks for all your great work!

Lithium73, 2021-04-18:
Hello to all, 
I made it work by replacing the moonlight-common-c submodule by the one mentioned above. 
Then : 
- re-build and install enet ( 7.0.5)
- follow compilation instructions on this repo.

dagonslair, 2021-04-20:
Same problem here.

I'm no programmer, so I don't know how to replace the moonlight-common-c submodule...

A new updated release would really great.
thanks !

Cageman1984, 2021-04-23:
Same problem here. I fixed it by rolling back to Driver 465.89 with the earlier version of GFE. (clean install)

Should we just apt-get update in a while to update moonlight embedded, or is replacing the moonlight-common-c submodule the only solution? In the last case... how do i do that?

dagonslair, 2021-04-24:
Anyone can help ?
If there's no apt-get update, how can we replace the moonlight-common-c submodule ?

Thanks !

Cageman1984, 2021-04-24:
I would like to know as well, but don't forget that rolling back to version 465.89 with GFE 3.21 is a fine solution for the meantime.

Cageman1984, 2021-04-26:
> I made it work by replacing the moonlight-common-c submodule by the one mentioned above.
> Then :
> 
> * re-build and install enet ( 7.0.5)
> * follow compilation instructions on this repo.

Last night I accidentally opened GFE for the third time. This automatically installs the update and therefore breaks my connection.  I have to uninstall gfe and install the old version again. 

Maybe it would be easier if I try to 'replace the moonlight-common-c submodule'. Could you give me a little hint direction in how to do so? I'm not an absolute pro in linux, but I can follow instructions and think a bit for myself ;)...

Vegz78, 2021-04-26:
> Last night I accidentally opened GFE for the third time. This automatically installs the update and therefore breaks my connection. I have to uninstall gfe and install the old version again.
>
> Maybe [...]

It finally worked for me by following the instructions here:
https://github.com/irtimmer/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC

If anyone gets the same segfaults/sigrestorer error e.g. when CMake/compiling on RPi OS buster, it worked for me to downgrade from gcc version 8 to 7...

Cageman1984, 2021-04-27:
> It finally worked for me by following the instructions here:
> https://github.com/irtimmer/moonlight-embedded/wiki/Full-details-for-Compilation-and-Installation-in-Raspbian-Or-OSMC
>
> If anyone gets the [...]

Thanks! That seems a pretty easy to follow guide, I'll try it soon. 

One question though. What is the effect of apt get update/upgrade on a self compiled program (such as moonlight in this case)? Does it have no effect at all and will I miss future updates? Does it simply keep my application up to date? Or does is break when I update?

Furthermore, why does an apt get update not update moonlight with the latest libraries? Do I need to add repositories, or do the developers of moonlight need to do something?

Vegz78, 2021-04-27:
@Cageman1984:

Compiling and manually installing yourself is totally independent of apt and must also be updated manually.

To avoid any chance of conflicts, I would recommend to uninstall the apt moonlight package before compiling and installing this one from GitHub. -And vice versa.

You could always uninstall the manual compiled version again if a new fixed official apt package has been released and you want automatic future updates again: https://stackoverflow.com/a/14516283/12802435

If moonlight embedded is not among the official Raspberry Pi OS apt packages, you could add this line to the file /etc/apt/sources OR in a new file under folder /etc/apt/sources.list.d: 
_deb http://archive.itimmer.nl/raspbian/moonlight buster main_
(Remember the correct OS version; buster, stretch, jesse etc.)

I don't know the specifics of making releases, either at GitHub or in official Raspberry repositories, but I believe this is an active action by the maintainers of this GitHub repository.

Cageman1984, 2021-04-27:
Thank you for the detailed info vegz78!

I used the apt version because I wanted to stay up to date automatically. If that is not the case, why use the apt version? I'll switch to compiling myself :).

Vegz78, 2021-04-27:
> Thank you for the detailed info vegz78!
> 
> I used the apt version because I wanted to stay up to date automatically. If that is not the case, why use the apt version? I'll switch to compiling myself :).

You are welcome and hope you get it to work, too!

The apt version is still to stay up to date automatically and effortlessly, and in addition, you get a better quality tested product as a part of an official release, not just a quick and work-in-progress fix/merge/update from GitHub(which very seldom, but sometimes don't work and can cause problems). Since the amazing Moonlight(which is both better than and saves us a lot of money compared to many of its commercial counterparts) is totally free and based only on community/spare time work, I guess the the former taking somewhat more time is understandable and appreciated.  ;-)

On the other hand, by compiling yourself, you get quicker and cutting edge updates, you learn a lot more, you can start tweaking certain things to your own liking, and suddenly you may even have new a product of your own and have learned to program a bit on the way! ;-)

Cageman1984, 2021-04-27:
I was able to get it to work when compiling moonlight myself! 

However, I've run in some new problems. First, no audio. Audio runs through my AV jack, not HDMI. Apparently you have to tell moonlight that by using -audio hw:1,0. After that it works, but it's really low volume! How can I increase the volume? The volume mixer seems to think I'm at 100%

second, my controller (Xbox one wireless controller, using the dongle and the XOW beta driver) is not recognized. It's not the biggest problem, I fall back to a default gamepad that works as well. Strange thing is though that Xbox one wireless controller is mentioned in gamecontrollerdb.txt, but under a different ID. Simply changing the ID does not seem to work...

dagonslair, 2021-05-08:
Any news about an apt-get update ?

Cageman1984, 2021-05-09:
We're still stuck on 2.4.11 (march 9). The windows version has recently been updated. Maybe moonlight embedded will follow soon?

I would say that purging the moonlight package and building it yourself is a decent solution for the moment. It's not nearly as difficult as it sounds. It's better dan rolling back GFE all the time.

I made two SD cards. One with the apt version and one with the self compiled version. Just check the moonlight embedded version on GitHub from time to time and switch when it's updated (that's my plan)

cgutman, 2021-05-09:
Since the APT official repo still hasn't been updated, I [wrote a set of packaging scripts](https://github.com/cgutman/moonlight-embedded-packaging) based on those I use to package Moonlight Qt and the old debian packaging files contained in the `raspbian/stretch` branch to create the package from the latest code myself.

You can download the build from the artifacts of https://ci.appveyor.com/project/cgutman/moonlight-embedded-packaging and install it with `dpkg -i`. You may need to run `apt install --fix-broken` to install dependencies. This should work to upgrade the official build from the APT repository too.

Run the following commands to install the updated build:
```
wget https://ci.appveyor.com/api/buildjobs/sop0mlityjthixsl/artifacts/moonlight-embedded_2.4.11-1buster1_armhf.deb
sudo dpkg -i moonlight-embedded_2.4.11-1buster1_armhf.deb
sudo apt install --fix-broken
```

ss-gnalvesteffer, 2021-06-13:
Confirmed that the workaround involving rolling back the graphics drivers to 465.89 w/ GFE 3.21.0.36 works.

Graphics Drivers: https://www.nvidia.com/en-us/drivers/results/172062/
GFE: https://us.download.nvidia.com/GFE/GFEClient/3.22.0.32/GeForce_Experience_v3.21.0.0.exe

1010101010101, 2021-06-18:
@ss-gnalvesteffer Thanks, the GFE downgrade did the trick! I didn't need to downgrade the Graphics drivers (466.27) on my end.

dagonslair, 2021-06-19:
compiling moonlight myself worked.
Thanks to Vegz78.

Oranjoose, 2021-07-08:
moonlight embedded is still not updated, and the newest version of GFE still doesn't work with it. If the maintainers of this project never update it, I'm guessing the chances of a new GFE version fixing this problem is very slim.

sander1095, 2021-07-09:
@irtimmer Could you shine some knowledge on this issue? I have currently built this app from source and used workarounds posted in this issue to fix this, but would you ever be able to provide some fixes or would you be willing and be able to accept PR's?

OneCricketeer, 2021-07-23:
Looks like there have been [recent commits to the C module](
https://github.com/moonlight-stream/moonlight-embedded/commit/d7f9bc0b25f744a4913c5b6198f4f96e3d85c8bd#diff-227655881fc8052373149ace0b1846c3ecfccb1c103cb76c616b76be1cf2b870) 👀

cgutman, 2021-07-26:
This is fixed and released in v2.5.0.

You'll need to [follow the steps here](https://github.com/moonlight-stream/moonlight-embedded/wiki/Packages) to install the new Cloudsmith repo to get the new Raspbian package.

closed 2021-07-26 by cgutman

Cageman1984, 2021-07-26:
Will try it soon!