moonlight-stream/moonlight-embedded #815: segfault during start stream

uabassguy, 2021-04-22:
Please provide the following info.

**_NVidia Geforce Experience version:_**
466.11
**Moonlight Embedded version:**
latest master 7c8795febc9fad98084aa10fc5918985cbd8081e
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
compiled from source
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Pi 4 
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
Raspbian Linux 5.4.51-v7l+ armv7l GNU/Linux

**Verbose output `-verbose` of Moonlight Embedded:**
Does not provide any useful info because of segfault, strace attached instead
[moonlight-segfault.txt](https://github.com/irtimmer/moonlight-embedded/files/6354736/moonlight-segfault.txt)

**What is the expected result?**
should work, stopped working in last couple days

**What happens instead of that?**
crashes before opening stream

Vegz78, 2021-05-15:
I had the same issue on one of my devices recently compiled, and segfault went away if I executed the bulit executable from inside the  ../moonlight-embedded/build folder. Just made a bash script that always changes to this folder and executes this file.

Not sure if this applies to you... ;-)

uabassguy, 2021-05-16:
huh, that makes sense. After I compiled the binary I moved it to /use/local/bin, then had this problem. I ended up using the one the retropie script installs in supplemental, that one works fine. Thanks @Vegz78

closed 2021-05-16 by uabassguy