moonlight-stream/moonlight-embedded #818: Unable to pair with host

guaycuru, 2021-05-07:
Please provide the following info.

**_NVidia Geforce Experience version:_**
3.22.0.32
**Moonlight Embedded version:**
latest master [7c8795f](https://github.com/irtimmer/moonlight-embedded/commit/7c8795febc9fad98084aa10fc5918985cbd8081e)
**Moonlight Embedded source:** _repository/included in distribution/compiled from source/..._
Compiled from source
**Moonlight Embedded running on:** _Raspberry Pi/Cubox-i/Hummingboard/Other linux device/..._
Raspberry Pi 3
**Moonlight Embedded running on distribution:** _Arch Linux/Raspbian/OpenELEC/..._
OSMC

**Verbose output `-verbose` of Moonlight Embedded:**

```
Moonlight Embedded 2.4.7 (PI;ALSA;PULSE;EMBEDDED)
Searching for server...
Connect to 192.168.16.8...
Request https://192.168.16.8:47984/serverinfo?uniqueid=REDACTED&uuid=REDACTED
Request http://192.168.16.8:47989/serverinfo?uniqueid=REDACTED&uuid=REDACTED
Response:
<?xml version="1.0" encoding="UTF-16"?><root protocol_version="0.1" query="serverinfo" status_code="200" status_message="OK"><AuthenticationType>1</AuthenticationType><ConnectionState></ConnectionState><CurrentClient>0</CurrentClient><GfeVersion>3.22.0.32</GfeVersion><GsVersion>gs_04_31_29754594</GsVersion><HttpsPort>47984</HttpsPort><LocalIP>REDACTED</LocalIP><LocalIPs><Address>REDACTED</Address></LocalIPs><LocalIPs><Address>REDACTED</Address></LocalIPs><LoginState>1</LoginState><MaxLumaPixelsH264>737235008</MaxLumaPixelsH264><MaxLumaPixelsHEVC>589622848</MaxLumaPixelsHEVC><Mode>0</Mode><PairStatus>0</PairStatus><ServerCapability>3895</ServerCapability><ServerCodecModeSupport>3843</ServerCodecModeSupport><ServerColorSpaceSupport>2</ServerColorSpaceSupport><SupportedDisplayMode><DisplayMode><Height>2160</Height><RefreshRate>60</RefreshRate><Width>3840</Width></DisplayMode><DisplayMode><Height>2160</Height><RefreshRate>30</RefreshRate><Width>3840</Width></DisplayMode><DisplayMode><Height>1080</Height><RefreshRate>60</RefreshRate><Width>1920</Width></DisplayMode><DisplayMode><Height>1080</Height><RefreshRate>30</RefreshRate><Width>1920</Width></DisplayMode><DisplayMode><Height>720</Height><RefreshRate>60</RefreshRate><Width>1280</Width></DisplayMode><DisplayMode><Height>720</Height><RefreshRate>30</RefreshRate><Width>1280</Width></DisplayMode></SupportedDisplayMode><accountId>REDACTED</accountId><appversion>7.1.431.0</appversion><currentgame>0</currentgame><gamelistid>REDACTED</gamelistid><gputype>NVIDIA GeForce GTX 1070</gputype><hostname>REDACTED</hostname><mac>REDACTED</mac><numofapps>0</numofapps><resyncSuccessful>0</resyncSuccessful><state></state><uniqueid>REDACTED</uniqueid></root>

Invalid data received from server: Problem with the local SSL certificate
```

Also, when trying to pair via Luna (the Kodi launcher) this is found in the logs, which might be useful:

```
2021-05-07 12:56:28.122 T:1416982752   ERROR: /home/osmc/.kodi/addons/script.module.urllib3/lib/urllib3/connectionpool.py:1004: InsecureRequestWarning: Unverified HTTPS request is being made. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
                                              InsecureRequestWarning,
2021-05-07 12:56:28.316 T:1416982752   ERROR: [script.luna]: <type 'exceptions.ValueError'> - 400 BAD REQUEST
```

**What is the expected result?**

Moonlight should be able to pair with host

**What happens instead of that?**

Moonlight is unable to pair with host

guaycuru, 2021-05-07:
So changing editing `/etc/ssl/openssl.cnf` to change `DEFAULT@SECLEVEL` from 2 to 0 according to [this issue](https://github.com/irtimmer/moonlight-embedded/issues/751) fixes my problem, but that seems like an undesirable workaround as I'm downgrading the default sec level for my osmc box from 2 to 0. Could this be resolved in another way?

cgutman, 2021-05-07:
Please post a Wireshark packet capture of the TCP port 47984 traffic (`tcp.port == 47984`). Something is probably using a certificate that OpenSSL doesn't like at SECLEVEL=2. It could be Moonlight if you haven't generated a new certificate since https://github.com/irtimmer/moonlight-embedded/commit/519a14a42786bafd267a215f671d4d73ccf7b13e or it could be GeForce Experience if that cert hasn't been regenerated in a while.

guaycuru, 2021-05-07:
There you go. Please let me know if you need anything else.

[moonlight.zip](https://github.com/irtimmer/moonlight-embedded/files/6443345/moonlight.zip)

guaycuru, 2021-05-07:
Yes, sorry, that first `verbose` capture was using an older version, but the pcap was generated with a newer version:

```
moonlight pair -verbose
Moonlight Embedded 2.4.10 (PI;ALSA;EMBEDDED)
Searching for server...
Connect to 192.168.16.8...
NVIDIA NVIDIA GeForce GTX 1070, GFE 3.22.0.32 (gs_04_31_29754594, 7.1.431.0)
Please enter the following PIN on the target PC: 2421
Failed to pair to server: Problem with the local SSL certificate
```

I'm running 2.4.11 (taken from http://archive.itimmer.nl/raspbian/moonlight ) which according to [this issue](https://github.com/irtimmer/moonlight-embedded/issues/794) is actually 2.4.11 for which the version was not updated.

I could also try with the latest version from master but for some reason it reports itself as 2.4.7.
Should I give that another try?

cgutman, 2021-05-07:
It's the version that generates the initial creds that matters, not the version that is actually trying to use them. If the first version of Moonlight you ran was some old build that didn't have 519a14a then the problem will persist on newer versions.

You need to wipe `~/.cache/moonlight` and pair again to get new creds generated.

If that still doesn't work, upload your certificate file `~/.cache/moonlight/client.pem` so we can tell if it truly is a SHA256 certificate.

guaycuru, 2021-05-07:
Yes, wiping `~/.cache/moonlight` did it, I was able to successfully pair now!  Thank you!

closed 2021-07-26 by cgutman