moonlight-stream/moonlight-embedded #819: Alsa buffer underrun - workaround/feature request

royka1, 2021-06-22:
I'm using moonlight with a Asus Tinker Board (armbian), every 20 minutes or so I get Alsa error's.
I've tried a lot of things to avoid the buffer underrun, but can't solve the problem, however I've found a workaround.
When there is a buffer underrun with Alsa, moonlight won't recover so you have to restart the whole stream, this I find quite anoying.
Now I start moonlight with the -local audio option and start a "scream" stream for the audio. 
https://github.com/duncanthrax/scream

With scream I also get some buffer underruns (about 5 in 2hrs), but it recovers from it, so you only notice a loss of audio then.

Is it possible to code moonlight this way?

AreaScout, 2021-07-22:
Why not using pulseaudio on top of alsa ? Buffer underruns are gone with it

royka1, 2021-07-22:
> Why not using pulseaudio on top of alsa ? Buffer underruns are gone with it

Tried that, but I had still Alsa errors, but instead Writei -2 I had Writei -11 if I remember correctly. And instead that the whole stream stopped, only the audio stopped. After that I've tried PipeWire, then the audio kept working, but at the moment it would normally stop, the audio was distorted. Playing with the buffer length didn't make a difference

AreaScout, 2021-07-22:
Those Writei -11 errors only happen with ALSA iirc, you have to build moonlight with pulseaudio support and start the daemon with _pulseaudio --start_